react-native-camera/react-native-camera #2222: RNCamera preview doesn't always resume when app comes to the foreground

thgaskell, 2019-04-21:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

- iOS

**Versions**

- iOS: `12.2 (on iPhone 6)`
- react-native-camera: `2.6.0`
- react-native: `0.59.5`
- react: `16.8.3`

**Description/Current Behaviour**

**Expected Behaviour**

1. Open camera.
2. Send the app to the background (Hit the home button).
3. Launch the app again.
4. Go to step 2.

Camera preview should **always** resume.

**Steps to Reproduce**

1. Open camera.
2. Send the app to the background (Hit the home button)
3. Launch the app again
4. Go to step 2.

After a few times the camera preview does not seem to resume when becoming active. Strangely, it just seems to go out of sync because if you close the app again and focus, it _will_ resume.

**Additionals**

This may be a duplicate of #2144?

I don't know much about the native implementation, but I came across #2158 which lead me to look at some of the native code.

My guess that it may be related to how `RNCamera` handles dispatching pausing and unpausing when the app comes to the foreground?

**Workaround**

I was able to work around this issue by listening to the `AppState` `change` event to manually call `resumePreview()` and `pausePreview()` which leads to consistent app behavior.

```tsx
  handleAppStateChange = (nextAppState: AppStateStatus) => {
    /* Reference to RNCamera instance */
    if (!this.camera) {
      return;
    }

    switch (nextAppState) {
      case "active":
        this.camera.resumePreview();
        break;
      case "background":
        this.camera.pausePreview();
        break;
      case "inactive":
      default:
    }
  };

  componentDidMount() {
    AppState.addEventListener("change", this.handleAppStateChange);
  }

  componentWillUnmount() {
    AppState.removeEventListener("change", this.handleAppStateChange);
  }
```

thgaskell, 2019-04-22:
After using the app with the workaround I would still occasionally run into the same issue. So I decided to take another look at the native code.

It appears that the issue is coming from this line in the `bridgeDidForeground` method:
https://github.com/react-native-community/react-native-camera/blob/1b9fb63e56368f90be2b94e32815e2837b53da41/ios/RN/RNCamera.m#L785

If I remove the line completely or set `self.paused = YES`, the camera the camera always resumes.

> The camera still appears to be paused in the App Switcher view.

My guess is that some sort of unintended side-effect is happening?

sibelius, 2019-04-22:
it looks like this is being there for a lot of time, maybe it is wrong

what should be the default behaviour?

thgaskell, 2019-04-22:
I think if the camera is open and put the app in the background, when you come back it should still be running. (That's the same behavior of the default `Camera` app.)

But I'm really not sure how `AVCaptureSession` is being used in `react-native-camera`?

This comment from a few years ago says you shouldn't need to pause the session, and it will automatically pause/resume when the app is put into the background: https://forums.developer.apple.com/thread/61406#173744

sibelius, 2019-04-22:
feel free to send a PR fixing this

closed 2019-06-28 by stale[bot]

gbalduzzi, 2020-01-27:
I'm experiencing the same issue. I even tried to remove the camera component when the app goes in background and restoring it when the app returns active, but occasionally it still freezes on a black screen and the only way to make it work again is reopening the app.

Does anyone know a feasable fix/workaround?

hushicai, 2020-03-11:
same problem here.
react-native-camera: 1.11.1
react-native:0.59.10

Asinox, 2020-06-05:
Here as well
"react-native": "0.61.5",
"react-native-camera": "^3.27.0",

fuhrthom, 2022-04-24:
Same problem happening to me.
react-native:0.64.3
expo-camera: ~12.1.2