react-native-camera/react-native-camera #2224: Tried to use permissions API but the host Activity doesn't implement PermissionAwareActivity.

icxcat, 2019-04-22:
Tried to use permissions API but the host Activity doesn't implement PermissionAwareActivity.

sibelius, 2019-04-22:
can you give more info? and more logs?

please use issue template

icxcat, 2019-04-22:
Hello，Sorry, I am using a mobile phone to edit.

I am using the RN version of 0.59.4. Manually add the camera to the project according to the documentation. iOS can run normally, but Android throws an error.  I have configured permissions in Android.

icxcat, 2019-04-22:
IOS
[image]

Android
[image]

icxcat, 2019-04-22:
```
b914c9eb-f5ef-420a-8bc4-4fcc6895e23d:67187 Possible Unhandled Promise Rejection (id: 0):
Error: Tried to use permissions API but the host Activity doesn't implement PermissionAwareActivity.
Error: Tried to use permissions API but the host Activity doesn't implement PermissionAwareActivity.
    at createErrorFromErrorData (blob:file:///b914c9eb-f5ef-420a-8bc4-4fcc6895e23d:2108:17)
    at blob:file:///b914c9eb-f5ef-420a-8bc4-4fcc6895e23d:2060:27
    at MessageQueue.__invokeCallback (blob:file:///b914c9eb-f5ef-420a-8bc4-4fcc6895e23d:2627:18)
    at blob:file:///b914c9eb-f5ef-420a-8bc4-4fcc6895e23d:2358:18
```

closed 2019-04-22 by icxcat

cipherGG, 2019-07-16:
do u fix this?

mattprivman, 2019-07-31:
To fix this, implement PermissionAwareActivity in the root ReactNative.java file. 

This [Stackoverflow answer](https://stackoverflow.com/a/45879293/2209974) provides a good example of what needs to be done.

helciofranco, 2019-10-11:
Any solution? This is happening just in Android.