react-native-camera/react-native-camera #2227: getting Frames from Camera

eranabir, 2019-04-23:
# Feature-Request

**Describe the Feature**
is there a way to receive the frames from the camera to the client ? 

**Possible Implementations**
i tried on IOS to get it from the captureOutput function in RNCamera.m
but no luck 

would love if it will be like that :

```javascript 

handleFrame=(frame)=>{console.log(frame)}

<RNCamera
      onFrame={this.handleFrame}/>

````

sibelius, 2019-04-23:
this would be slow

eranabir, 2019-04-23:
i can live with it :)

c1ngular, 2019-04-24:
how about getting the video stream ?

hartatovich, 2019-04-24:
> this would be slow

why this would be slow if the frames coming from the native code ?

the idea here is to get the frames in native code and then just send them to the js or am i missing something here ?

from what i understand if we break the frames in the js it self then everything will become super slow..

would love to hear your thoughts @sibelius

sibelius, 2019-04-24:
passing big chunks of data in the bridge is slow

you can implement on a PR and we can test more

closed 2019-06-30 by stale[bot]

tvhieuit, 2019-07-04:
How to use RNCamera with posenet.
https://github.com/jscriptcoder/tfjs-posenet