react-native-camera/react-native-camera #2230: Create Camera Singleton

ehorodyski, 2019-04-24:
# Question

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
I'm working on an app that uses a non-standard navigation UI (see picture attached), where the navigation is split into 3 buttons:

1. Help
2. Take Picture (or navigate to camera if on other screens)
3. About

I'd like to use the middle icon to take a picture, however, that button is part of my navigation component, not the screen in which I plan to place the RNCamera instance. 

My question is, how I can I either pass the `ref` that needs to be created within the RNCamera component to the navigation component to call it's `takePicture` method, or better yet, create a singleton instance of the camera so that my navigation can call a business-logic layer? Something like this:

CameraComponent
```
...
render{
    return (
        <RNCamera
          ref={(ref) => { this.camera = ref; }}
         ...
    );
}
...
```

NavCameraButtonComponent
```
...
onPress() {
    // Business-layer logic to take picture and manipulate to my desire
    CameraSingleton.takePictureAsync();
}
...
```

**Tags**
Help, Guide

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-04-24:
Are you using some navigation on the screens? Or you just wrap your app around a single component?

ehorodyski, 2019-04-24:
I'm using react-navigation. 

I'm pretty new to React/React-Native so I apologize if I made any of this confusing.

MateusAndrade, 2019-04-24:
> I'm pretty new to React/React-Native so I apologize if I made any of this confusing.

I use react-native/navigation since the last year but i steel fell a newbie. hahahaha

Btw, i got on a similar use case: when the user enters on a specific screen, i started the RNCamera instance, set the ref to the component to use this on a button, etc.

On your case, you could write a function that accepts the response from RNCamera, and returns this data, as a promise for example. I think writing a component that encapsulates RNCamera, and exports the data when it captures a picture would be a better option.

The only problem that i see on your case, is if you go to your camera stack(if you are using stacks on react-navigation) and then switch to another tab, if the camera will unmount our stay open. Because keeping the camera active can cause some performance issues.

Hope it helps!

ehorodyski, 2019-04-24:
@MateusAndrade -- I'm still a bit confused, this part of RNCamera:

```
<RNCamera 
    ref={(ref) => { this.camera = ref; }} 
    ... 
    />
```

Scopes the camera reference to the Component it's being composed in. How would I pass `ref` to another component?

What I suppose I could do is build a CameraManager of sorts:

```
import { CameraManager } from './camera.manager.js';

export default class CameraScreen extends Component {
    render() {
        return (
            <RNCamera 
                ref={(ref) => { CameraManager.camera = ref }}
             />
        );
    }
}
```

Then in my other component:

```
onButtonPress() {
    CameraManager.camera.takePictureAsync();
}
```

MateusAndrade, 2019-04-24:
> Scopes the camera reference to the Component it's being composed in. How would I pass `ref` to another component?

> ```
> onButtonPress() {
>     CameraManager.camera.takePictureAsync();
> }
> ```

You can try pass the ref though props to a father/children. But by what i understand, you want to call the ref to take a picture, when you click on the onButtonPress. Thats right?

If you are trying to pass this ref trough components that are father/children you can pass it as a prop. But i dont think it would possible doing something similar to a Singleton. Maybe, you can call a modal and init the camera on this component

ehorodyski, 2019-04-24:
Yeah, unfortunately I'm looking to have the parent component call `takePictureAsync`, but the child component is the one that init's RNCamera. I'll send some better code over tomorrow once I'm around my dev machine. But the idea is either:

1. Pass the `ref` up to the parent component, so the parent component can take the picture, or;
2. Have the child initialize the camera, store `ref` in a shared class, and let the parent call `takePictureAsync` from that shared instance..

sibelius, 2019-04-24:
useEffect

ehorodyski, 2019-04-24:
@sibelius - Do you mind sharing a link or code sample? I'm unsure what library to use for effects w/ Redux.

MateusAndrade, 2019-04-24:
You can use react-redux or even react-hooks.

ehorodyski, 2019-04-24:
I guess the issue would still remain -- where do I get the reference to the Camera in a parent component?

sibelius, 2019-04-24:
`ref={ref => this.props.cameraRef(ref)}`

ehorodyski, 2019-04-24:
I've updated my initial post to hopefully clear up some confusion and link to the project. 

@sibelius - That was the approach I was thinking, but isn't a navigation view a child component of the navigation component? If that's the case, I still don't understand how I'd pass it up to it's parent.

MateusAndrade, 2019-04-25:
@ehorodyski you can dispatch a Action on your button press component. Listen tô this change on your camera component, and trigger the capture on the componente. I have fone something similar to It, but i can implement/fork your repo at the moment.

Tomorrow,  when i got my dev machine i will test onde thing on a fork and pass the example for you. :)

MateusAndrade, 2019-04-25:
@ehorodyski  you can do something like this:

```javascript
class CircleNavButton extends Component {

 // Dispatch your action
  takePicture = () => {
    console.log('takePicture', this.props);
  };

  render() {
    return (
      <TouchableOpacity onPress={this.takePicture} style={styles.circleNavButton}>
        <Icon name="camera" color={Colors.White} size={32} />
      </TouchableOpacity>
    );
  }
}

export default connect()(CircleNavButton);
```

On `takePicture` you will need to dispatch a action, a TAKE_PICTURE for example. And on your component `CameraScreen` you will need to listen to this change using the props or with another option( maybe a observable?). 

But i cant figure out how you would pass the Camera ref as you are suggesting. 

Also, you should give a try to [redux-thunk](https://github.com/reduxjs/redux-thunk) to dispatch actions, etc.

Btw, hope it heps!

ehorodyski, 2019-04-25:
It looks like it was just as easy as creating a static class to just store the camera in:

`camera.service.js`
```
export default class CameraService {
  static camera;

  static async takePicture() {
    if (this.camera) {
      const options = { quality: 0.5, base64: true };
      const data = await this.camera.takePictureAsync(options);
      console.log(data);
    }
  }
}
```

`CameraScreen.js`
```
...
import { CameraService } from '../../services';
...
ref={(ref) => { CameraService.camera = ref; }}
...
```

`CircleNavButton.js`
```
...
import { CameraService } from '../services';
...
<Icon
    name="camera"
    color={Colors.White}
    size={32}
    onPress={() => CameraService.takePicture()}
/>
...
```

As an aside; I think it would be extremely helpful to split the view component `<RNCamera>` from its native functionality. I would make `ref` have to be of the type of this functionality-driven class. Kind of split it into `RNCameraComponent` that has a `ref` that takes a type of `RNCamera`.

This would allow other components to consume parts that are currently exclusive to the `RNCamera` component without having to do a roundabout solution like I did. Conversely, the `<RNCamera>` component would only be responsible for displaying what the camera sees.

Additionally, some of the more compelling features of RNCamera, such as face-detection and barcode-scanning, could be abstracted out of the view so users could use it on already taken images, if that's not available at the moment.

closed 2019-04-25 by ehorodyski