react-native-camera/react-native-camera #2231: Camera does not auto focus or manual focus

mikerodham, 2019-04-24:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android: N/A
- iOS: 12.1.4
- react-native-camera: ^2.1.1
- react-native: 0.57.7
- react: 16.6.1
  

**Description/Current Behaviour**

I'm trying to read a Code39 and the camera will not focus on the code I am using. It simply stays focused on the background, if I cover the camera and show it only the card, it still does not focus and will not scan the code.

QR Codes work perfectly.

**Expected Behaviour**

I expect the camera to focus on the card, scan the code.

**Steps to Reproduce**

Try to scan a Code39.

I do not use Expo.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

mikerodham, 2019-04-26:
I found some code that seems relevant: (I don't know swift/obj c)
275-300
```
- (void)updateFocusDepth
{
    AVCaptureDevice *device = [self.videoCaptureDeviceInput device];
    NSError *error = nil;

    if (device == nil || self.autoFocus < 0 || device.focusMode != RNCameraAutoFocusOff || device.position == RNCameraTypeFront) {
        return;
    }

    if (![device respondsToSelector:@selector(isLockingFocusWithCustomLensPositionSupported)] || ![device isLockingFocusWithCustomLensPositionSupported]) {
        RCTLogWarn(@"%s: Setting focusDepth isn't supported for this camera device", __func__);
        return;
    }

    if (![device lockForConfiguration:&error]) {
        if (error) {
            RCTLogError(@"%s: %@", __func__, error);
        }
        return;
    }

    __weak __typeof__(device) weakDevice = device;
    [device setFocusModeLockedWithLensPosition:self.focusDepth completionHandler:^(CMTime syncTime) {
        [weakDevice unlockForConfiguration];
    }];
}
```

More specifically just this section here: 

```
    if (device == nil || self.autoFocus < 0 || device.focusMode != RNCameraAutoFocusOff || device.position == RNCameraTypeFront) {
        return;
    }
```

fabriziobertoglio1987, 2019-05-04:
I gave the [following solution on stackoverflow](https://stackoverflow.com/a/55966437/7295772)

IOS has [three `Focus Modes`](https://developer.apple.com/library/archive/documentation/AudioVideo/Conceptual/AVFoundationPG/Articles/04_MediaCapture.html#//apple_ref/doc/uid/TP40010188-CH5-SW2#Focus_Modes). You need to use `AVCaptureFocusModeContinuousAutoFocus`

>[AVCaptureFocusModeContinuousAutoFocus](https://developer.apple.com/documentation/avfoundation/avcapturefocusmode/avcapturefocusmodecontinuousautofocus): The camera continuously autofocuses as needed.

>**You use the [isFocusModeSupported](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1390215-isfocusmodesupported): method to determine whether a device supports a given `focus mode`**, then set the mode using the focusMode property.

`react-native-camera` will change the focus in two different scenarios (you can set a breakpoint with `xcode` on this lines):

1. [`focusWithMode`](https://github.com/react-native-community/react-native-camera/blob/960c5b1c1687742bcadd570f676e2689339e48b0/ios/RCT/RCTCameraManager.m#L1054) method will set the focus only if your frontcamera supports `isFocusPointOfInterestSupported` and `AVCaptureFocusModeContinuousAutoFocus`

The method will change the focus mode `[device setFocusMode:focusMode];` to `AVCaptureFocusModeContinuousAutoFocus` only if the following condition is returns `true`

    [device isFocusPointOfInterestSupported] && [device isFocusModeSupported:focusMode]

If the condition returns `false`, there is no `autofocus`, **but** the picture may [be focused on](https://stackoverflow.com/a/17226418/7295772) ExposureMode [`[device setExposureMode:exposureMode];`](https://github.com/react-native-community/react-native-camera/blob/960c5b1c1687742bcadd570f676e2689339e48b0/ios/RCT/RCTCameraManager.m#L1059)

2. [`updateAutoFocusPointOfInterest`](https://github.com/react-native-community/react-native-camera/blob/1b9fb63e56368f90be2b94e32815e2837b53da41/ios/RN/RNCamera.m#L240) changes the focus when the user taps on the screen, based on the `x, y` coordinates of his touch.

As there are several posts ([post 1](https://stackoverflow.com/questions/17219049/iphone-5-front-camera-tap-to-focus), [post 2](https://stackoverflow.com/questions/18618229/autofocus-in-iphone-4s-front-camera-adjustingfocus-observer-not-being-invoked), [post 3](https://stackoverflow.com/questions/17219049/iphone-5-front-camera-tap-to-focus), [post 4](https://stackoverflow.com/questions/9729898/iphone-front-camera-tap-to-focus)) on stackoverflow stating that different iphone versions do not support all type of autofocus with the frontcamera, I suggest you to **set breakpoints** on those line of codes and check the value of [`isFocusModeSupported`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1390215-isfocusmodesupported) and [`isFocusPointOfInterestSupported`](https://developer.apple.com/documentation/avfoundation/avcapturedevice/1390436-isfocuspointofinterestsupported)

closed 2019-05-09 by mikerodham