react-native-camera/react-native-camera #2232: asyncRecord returns a valid uri, but file is null after Java update on Android

fostahgix, 2019-04-24:
# Question

**To Do First**
- [ x ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ x ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ x ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Env: 
Android 8 (Oreo)
```"react-native": "0.57.8",
"react-native-camera": "^2.6.0", (also tried the git install)
```
```
$ javac -version
javac 1.8.0_191
```
After upgrading Java on my Mac to Java 8 Update 211, for some reason on Android I am getting null files. I am not having any issues on iOS. Before the Java update, everything was fine.

Using `asyncRecord`, I get no errors, and I get back an object like 
` { uri: "file:///data/user/0/com.myapp/cache/Camera/13dd8af5-e2ea-4a3b-bf60-0616f189afe2.mp4", videoOrientation: 1, deviceOrientation: 1, isRecordingInterrupted: false}
`
Using `react-native-fs`, I checked if the file exists, and it returns `true`. However,  when I try to read the file size it comes back as null. Eg 
`Attempt to get length of null array`

I also checked my filesystem inside android and I do not see the files, eventhough react-native-fs says the file uri does exist.

Here is my code that was working before the Java update:
```
const options = {
      quality: RNCamera.Constants.VideoQuality['480p'],
      orientation: 'portrait',
      maxDuration: 4,
    }
    const { uri } = await this.camera.recordAsync(options)
    const result = await RNFS.exists(uri)
```
I was previously previewing the recorded video with `react-native-video`, but now it is throwing an error unknown format(I believe due to a null file)

This was all working fine without any issues and I haven't made any changes other than upgrading Java. 

Anyone have any ideas? Thanks!

**Tags**
Java, react-native-camera, react-native, android, android 8

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

fostahgix, 2019-04-26:
I have found the issue, however it is not clear as to what the root cause was.

**The issue:**
Inside my async class method, I was calling `setState` before `await asyncRecord`, and after the await, I was calling another method to update state.

**Eg.**
```
  triggerRecording = () => {
    this.setState(
      { recording: true, processing: false, preview: false, uri: null, codec: null },
      this.startRecording
    )
  }

  startRecording = () => {
      const options = {
        quality: RNCamera.Constants.VideoQuality['480p'],
        orientation: 'portrait',
        maxDuration: 4,
      }
      if (Platform.OS === 'ios') options.codec = RNCamera.Constants.VideoCodec.H264
      const { uri, codec = 'mp4' } = await this.camera.recordAsync(options)

      this.previewVideo(uri, codec)

  }

  previewVideo = async (uri, codec) => {
    this.setState({
      recording: false,
      processing: false,
      preview: true,
      uri,
      codec,
    })
  }
```
**The Fix/Workaround**
By wrapping start recording in a setTimeout it resolved the issue. [Found a similar issue/not sure if its related ](https://github.com/react-native-community/react-native-camera/issues/1967). If anyone has some insight, or if I am doing this wrong please let me know. Thanks!

```
  startRecording = () => {
    setTimeout(async () => {
      const options = {
        quality: RNCamera.Constants.VideoQuality['480p'],
        orientation: 'portrait',
        maxDuration: 4,
      }
      // set codec to h264 for compatibility witha android
      if (Platform.OS === 'ios') options.codec = RNCamera.Constants.VideoCodec.H264
      // default to mp4 for android as codec is not set
      const { uri, codec = 'mp4' } = await this.camera.recordAsync(options)
      // / Preview

      this.previewVideo(uri, codec)
    }, 1)

    // this.setState({ processing: false })
  }
```

fostahgix, 2019-05-01:
So, it appears after updating from the latest 1.x version to 2.x this issue has gone away. I will go ahead and close this issue, but I'm still not sure why a setState would cause an issue.

closed 2019-05-01 by fostahgix

MrPluto, 2019-05-21:
@fostahgix hi man, i have the same issue, when i finish record,i can play the video by using the uri as an video source,but i cant get any file in my phone.

traxx10, 2019-11-20:
I still have this issue, resolved it with a settimeout