react-native-camera/react-native-camera #2233: Error when running example for android

muhammadridho, 2019-04-25:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
Androdi

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**
I'm trying to run example code for android. but when code is compiled i get this error from android
[image]

**Expected Behaviour**
i am expected this example run

**Steps to Reproduce**
try react-native link,  and reinstall modules

arisupriatna14, 2019-04-25:
Close your metrobundler, and try build again

muhammadridho, 2019-04-25:
haha, i've just solved before

closed 2019-04-25 by muhammadridho