react-native-camera/react-native-camera #2234: Build APK Android Failed

arisupriatna14, 2019-04-25:
# Bug Report

**To Do First**

- [✔️ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

Android

* What went wrong:
Execution failed for task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'.
   > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-vision:19.0.3
      > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
         > Failed to list versions for com.google.firebase:firebase-ml-common.
            > Unable to load Maven meta-data from https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-common/maven-metadata.xml.
               > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-common/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
         > Failed to list versions for com.google.firebase:firebase-ml-common.
            > Unable to load Maven meta-data from https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml.
               > Could not HEAD 'https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
         > Failed to list versions for com.google.firebase:firebase-ml-common.
            > Unable to load Maven meta-data from https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-common/maven-metadata.xml.
               > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/firebase/firebase-ml-common/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
         > Failed to list versions for com.google.firebase:firebase-ml-common.
            > Unable to load Maven meta-data from https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml.
               > Could not HEAD 'https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml'.
                  > Connection reset
   > Could not resolve com.google.android.gms:play-services-vision:[17.0.2].
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-vision-face-model:17.0.2
      > Could not resolve com.google.android.gms:play-services-vision:[17.0.2].
         > Failed to list versions for com.google.android.gms:play-services-vision.
            > Unable to load Maven meta-data from https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/maven-metadata.xml.
               > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.android.gms:play-services-vision:[17.0.2].
         > Failed to list versions for com.google.android.gms:play-services-vision.
            > Unable to load Maven meta-data from https://maven.google.com/com/google/android/gms/play-services-vision/maven-metadata.xml.
               > Could not HEAD 'https://maven.google.com/com/google/android/gms/play-services-vision/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.android.gms:play-services-vision:[17.0.2].
         > Failed to list versions for com.google.android.gms:play-services-vision.
            > Unable to load Maven meta-data from https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/maven-metadata.xml.
               > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-vision/maven-metadata.xml'.
                  > Connection reset
      > Could not resolve com.google.android.gms:play-services-vision:[17.0.2].
         > Failed to list versions for com.google.android.gms:play-services-vision.
            > Unable to load Maven meta-data from https://maven.google.com/com/google/android/gms/play-services-vision/maven-metadata.xml.
               > Could not HEAD 'https://maven.google.com/com/google/android/gms/play-services-vision/maven-metadata.xml'.
                  > Connection reset

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

KemikalGeneral, 2019-06-07:
@arisupriatna14 
Hi, did you ever solve this error?
I'm using this camera which works fine running on a device, but now that I'm trying to build the APK, it's throwing an error:

> Execution failed for task ':app:lintVitalRelease'.
> > Could not resolve all artifacts for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'.
>    > Could not find com.google.firebase:firebase-ml-vision:12.0.1.

and

Could not find com.google.firebase:firebase-ml-vision-face-model:12.0.1.

Thanks!

guildeassis, 2019-06-21:
I also have this problem in version 2.10.2. 
For now I'm using version 1.13.1 and works.

aryanbhasin, 2019-07-06:
I have the same problem and haven't been able to find a solution so far

closed 2019-09-14 by stale[bot]

Andriiklymiuk, 2019-10-01:
This still isn't fixed in rn 0.61.1 , 
any help?

KemikalGeneral, 2019-10-01:
@Andriiklymiuk Due to the lack of help, I switched to this package which works better, has more features, and less setup. Never looked back since.

https://github.com/ivpusic/react-native-image-crop-picker

Andriiklymiuk, 2019-10-01:
@KemikalGeneral Thanks, I'll try it out)

Schabaani, 2019-12-15:
I had this issue. I changed repositories order in `android\build.gradle`

```
repositories {
        google()
        jcenter()

        mavenLocal()
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url("$rootDir/../node_modules/react-native/android")
        }
        maven {
            // Android JSC is installed from npm
            url("$rootDir/../node_modules/jsc-android/dist")
        }
        maven { url 'https://jitpack.io' }
   }
```

FansPro, 2020-06-22:
In my case,  I delete android/app/build and then rebuild will be ok

JoseIgnacio92, 2020-09-08:
Hi, in my case, I had to delete de gradle cache folder content

C:\Users\jiperez\.gradle\caches

then rebuild

fengandrain, 2020-12-22:
I solved my problem by [https://maven.aliyun.com/mvn/guide]。Don't thanks me,thanks alibaba