react-native-camera/react-native-camera #2235: While capturing video, Android freeze if home/blocked/minimized

cristianoccazinsp, 2019-04-25:
# Bug Report

Android

**Versions**

- Android: 8.1 (Moto G5), 9 (Pixel 2)
- react-native-camera: 2.2.2
- react-native: 0.59.4
- react: 16.8.6
- react-navigation: 3.6.0

While capturing a video, even if stopRecording() is called before the app transitions to background/inactive, the app will totally freeze after resuming.

I'm starting the video capture as: ```result = await this.camera.recordAsync(options);```

Then, on the app state change events, if background or inactive, I call ```stopRecording()```

I can get a console print right after the stopRecording call; however, nothing ever outputs after the await call; looks like the promise never resolves.

These are the logs that appear after resuming. No error, no crash, just this.

```
04-25 10:30:53.159 30414-30420/com.zinspector_rn I/zygote: Thread[3,tid=30420,WaitingInMainSignalCatcherLoop,Thread*=0xab7b6200,peer=0x13380120,"Signal Catcher"]: reacting to signal 3
04-25 10:30:53.553 30414-30420/com.zinspector_rn I/zygote: Wrote stack traces to '[tombstoned]'
04-25 10:32:02.787 30414-30420/com.zinspector_rn I/zygote: Thread[3,tid=30420,WaitingInMainSignalCatcherLoop,Thread*=0xab7b6200,peer=0x13380120,"Signal Catcher"]: reacting to signal 3
04-25 10:32:03.076 30414-30420/com.zinspector_rn I/zygote: Wrote stack traces to '[tombstoned]'
```

cristianoccazinsp, 2019-05-03:
Some more logs. Also, this happens even if the device rotates.

 W/CameraSource: Timed out waiting for incoming camera video frames: 479533666669 us

cristianoccazinsp, 2019-05-08:
Crash report from bugsnag

ANR Application did not respond for at least 5000 ms 
    MediaRecorder.java:-2 android.media.MediaRecorder.stop
    Camera1.java:1034 com.google.android.cameraview.Camera1.stopMediaRecorder
    Camera1.java:523 com.google.android.cameraview.Camera1.stopRecording
    CameraView.java:557 com.google.android.cameraview.CameraView.stopRecording
    CameraModule.java:313 org.reactnative.camera.CameraModule$7.execute
    UIViewOperationQueue.java:583 com.facebook.react.uimanager.UIViewOperationQueue$UIBlockOperation.execute
    UIViewOperationQueue.java:917 com.facebook.react.uimanager.UIViewOperationQueue$1.run
    UIViewOperationQueue.java:1025 com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches
    UIViewOperationQueue.java:46 com.facebook.react.uimanager.UIViewOperationQueue.access$2600
    UIViewOperationQueue.java:1085 com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded
    GuardedFrameCallback.java:29 com.facebook.react.uimanager.GuardedFrameCallback.doFrame
    ReactChoreographer.java:166 com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame
    ChoreographerCompat.java:84 com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame
    Choreographer.java:947 android.view.Choreographer$CallbackRecord.run
    Choreographer.java:761 android.view.Choreographer.doCallbacks
    Choreographer.java:693 android.view.Choreographer.doFrame
    Choreographer.java:935 android.view.Choreographer$FrameDisplayEventReceiver.run
    Handler.java:873 android.os.Handler.handleCallback
    Handler.java:99 android.os.Handler.dispatchMessage
    Looper.java:193 android.os.Looper.loop
    ActivityThread.java:6718 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:493 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:858 com.android.internal.os.ZygoteInit.main

cristianoccazinsp, 2019-05-10:
Would also like to add that this is suspiciously related to "stock" android; that is, Google Pixel and motorola phones. This doesn't seem to happen on samsung, nor xiaomi.

closed 2019-07-16 by stale[bot]

nachumFreedman, 2019-08-18:
This issue still exists

cristianoccazinsp, 2019-09-05:
Can we reopen this? This is still happening on Pixel 2 with Android 10. While recording, even if stop is called on app suspend, the app will just freeze.

Stack trace:

```
ANR  Input dispatching timed out (Waiting to send non-key event because the touched window has not finished processing certain input events that were delivered to it over 500.0ms ago.  Wait queue length: 15.  Wait queue head age: 16366.3ms.) 
    MediaRecorder.java:-2 android.media.MediaRecorder.stop
    Camera1.java:168 com.google.android.cameraview.Camera1.stop
    CameraView.java:308 com.google.android.cameraview.CameraView.stop
    RNCameraView.java:496 org.reactnative.camera.RNCameraView.onHostPause
    ReactContext.java:236 com.facebook.react.bridge.ReactContext.onHostPause
    ReactInstanceManager.java:674 com.facebook.react.ReactInstanceManager.moveToBeforeResumeLifecycleState
    ReactInstanceManager.java:498 com.facebook.react.ReactInstanceManager.onHostPause
    ReactInstanceManager.java:517 com.facebook.react.ReactInstanceManager.onHostPause
    ReactActivityDelegate.java:99 com.facebook.react.ReactActivityDelegate.onPause
    ReactActivity.java:58 com.facebook.react.ReactActivity.onPause
    Activity.java:7337 android.app.Activity.performPause
    Instrumentation.java:1465 android.app.Instrumentation.callActivityOnPause
    ActivityThread.java:4059 android.app.ActivityThread.performPauseActivityIfNeeded
    ActivityThread.java:4024 android.app.ActivityThread.performPauseActivity
    ActivityThread.java:3976 android.app.ActivityThread.handlePauseActivity
    PauseActivityItem.java:45 android.app.servertransaction.PauseActivityItem.execute
    TransactionExecutor.java:145 android.app.servertransaction.TransactionExecutor.executeLifecycleState
    TransactionExecutor.java:70 android.app.servertransaction.TransactionExecutor.execute
    ActivityThread.java:1816 android.app.ActivityThread$H.handleMessage
    Handler.java:106 android.os.Handler.dispatchMessage
    Looper.java:193 android.os.Looper.loop
    ActivityThread.java:6718 android.app.ActivityThread.main
    Method.java:-2 java.lang.reflect.Method.invoke
    RuntimeInit.java:493 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run
    ZygoteInit.java:858 com.android.internal.os.ZygoteInit.main
```

cristianoccazinsp, 2019-09-05:
Pull request here: https://github.com/react-native-community/react-native-camera/pull/2467

n1ru4l, 2019-09-05:
:tada: This issue has been resolved in version 3.3.3 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v3.3.3)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: