react-native-camera/react-native-camera #2239: How to save pictures/videos in a different location?

cristianoccazinsp, 2019-04-26:
Is it possible to save pictures in the app file storage rather than the cache folder?

For videos, there's a path parameter. However, this is only to override an existing video. Is it possible to give it a folder instead? I have tried using a folder but the app will just crash (cause it's not a file).

emilemoureau, 2019-04-28:
Yesss!

`https://github.com/itinance/react-native-fs`

For example:
```javascript
var { TemporaryDirectoryPath } = require('react-native-fs');

const options = { path: TemporaryDirectoryPath + uuidv1() + "video.mp4" }
await this.camera.recordAsync(options)
```

To save the video in a folder you just have to create it before with react-native-fs!

cristianoccazinsp, 2019-04-28:
What about pictures? Are there any technical constraints that made react-native-camera only save pictures into one location?

Also, can we really assume all devices record in mp4? Why can't the library just take a folder path instead  while still being responsible of the file name?

emilemoureau, 2019-04-28:
It also works with pictures! I have no problem on Android and iOS with the .mp4, I think it would make the process more complicated, but it's very easy to create a folder and put the images and video in it with react-native-fs!

cristianoccazinsp, 2019-04-28:
@emilemoureau  I see no "path" parameter for pictures.

I know that it would be easy to simply move the files from one folder to another afterwards, but I'm just trying to understand why can't react-native-camera capture and store into other folders right away. The copy process would definitely slow things and is not worth it if you are trying to implement a fast capture app.

MateusAndrade, 2019-04-28:
> @emilemoureau I see no "path" parameter for pictures.
>
> I know that it would be easy to simply move the files from one folder to another afterwards, but I'm just trying to understand why can't [...]

Maybe its because rn-camera saves the files to the temp folder inside the application. The process of saving the files to another location doenst see do be a problem using react-native-fs. Adding that feature to save the files to a desired location can make the things a little bit complicated on some cases, since the library relies on contributors to make their maintenance/add new features. 

But, new prs are always a good adition!

MateusAndrade, 2019-04-28:
Since the answer to the question has been answered, im closing the issue.

closed 2019-04-28 by MateusAndrade

arunrajarao, 2020-07-26:
is there  way to specify file name , I only see path name

roycedot, 2020-10-20:
fwiw, I patched my code

android/src/main/java/org/reactnative/camera/utils/ScopedContext.java

```
    public void createCacheDirectory(Context context) {
        //cacheDirectory = new File(context.getCacheDir() + "/Camera/");
        cacheDirectory = Environment.getExternalStoragePublicDirectory(Environment.DIRECTORY_PICTURES);
    }
```

This saves to **/Internal Storage/Pictures** for me (Android 10, Galaxy S9 w/an external SD card)

I needed to request WRITE_EXTERNAL_STORAGE permission

We did this because we were having problems creating the cache dir subdirectory (/Camera) on a device with no external SD card and low storage space.

saddamhussaints, 2021-08-25:
can someone provide code for changing path of clicked images...