react-native-camera/react-native-camera #2240: 【Android】Conflict with `react-native-device-info`, so difficult!

12343954, 2019-04-27:
# Bug Report

**To Do First**

- [✅ ] Did you try latest release?
- [✅ ] Did you try master?
- [✅ ] Did you look for existing matching issues?

**Platforms**

**Versions**

iOS: is ok
- Android:  ✅ 
- Android Studio: 3.1.4 & 3.4
- com.android.tools.build:gradle:3.3.1 & 3.4.0

- react-native-camera: 2.4.0 & 2.6.0
- react-native: 0.56.0
- react: 16.4.1

```
ext {
    minSdkVersion               = 19

    compileSdkVersion           = 28
    targetSdkVersion            = 28
    buildToolsVersion           = "28.0.3"
    supportLibVersion           = "28.0.0"

    googlePlayServicesVersion   = "16.1.0"
    googlePlayServicesVisionVersion = "15.0.2"
}

```

**Steps to Reproduce**

after installing manually this lib, 
1. without step 7 , because i don't want to use 'google service: gms, firebase'.
2. without 'com.google.android.gms', 
3. without 'com.google.firebase', 

‘Sync now’ was stopped by this errors
some like these(i tried many many solutions, and couldn't fix)：

1. org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':react-native-camera:compileGeneralDebugJavaWithJavac'

superp, 2019-04-30:
I have the same problem :(

        buildscript {
        ext {
            buildToolsVersion = "28.0.3"
            minSdkVersion = 19
            compileSdkVersion = 28
            targetSdkVersion = 28
            supportLibVersion = "28.0.0"
        }
        repositories {
            google()
            jcenter()
        }
        dependencies {
            classpath 'com.android.tools.build:gradle:3.3.0'

            // NOTE: Do not place your application dependencies here; they belong
            // in the individual module build.gradle files
        }
    }

My dependencies:

        dependencies {
        implementation 'com.android.support:support-v4:28.0.0'
        implementation project(':react-native-camera')
        implementation project(':react-native-config')
        implementation project(':react-native-device-info')
        implementation project(':react-native-i18n')
        implementation project(':react-native-gesture-handler')
        implementation fileTree(dir: "libs", include: ["*.jar"])
        implementation "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
        implementation "com.facebook.react:react-native:+"  // From node_modules
        implementation 'com.google.android.material:material:1.0.0'
    }
Here is my log:

        > Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/RNCameraView.java:10: error: package android.support.v4.content does not exist
    import android.support.v4.content.ContextCompat;
                                     ^
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:27: error: package android.support.annotation does not exist
    import android.support.annotation.IntDef;
                                     ^
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:28: error: package android.support.annotation does not exist
    import android.support.annotation.NonNull;
                                     ^
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:29: error: package android.support.annotation does not exist
    import android.support.annotation.Nullable;
                                     ^
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:30: error: cannot find symbol
    import android.support.v4.os.ParcelableCompat;
                                ^
      symbol:   class ParcelableCompat
      location: package android.support.v4.os
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:31: error: cannot find symbol
    import android.support.v4.os.ParcelableCompatCreatorCallbacks;
                                ^
      symbol:   class ParcelableCompatCreatorCallbacks
      location: package android.support.v4.os
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/CameraView.java:32: error: package android.support.v4.view does not exist
    import android.support.v4.view.ViewCompat;
                                  ^
    /Users/developer/www/QRCodeScanner/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/AspectRatio.java:21: error: package android.support.annotation does not exist
    import android.support.annotation.NonNull;
    ...

superp, 2019-05-01:
After more 10 hours trying to fix this, I found a solution.
In my project I use androindx:

    # gradle.properties
    android.useAndroidX=true
    android.enableJetifier=true

So, I had to change all old support.v4 code in the third party libs, after my app had been migrated to AndroidX, for example:

    android.support.v4.content.ContextCompat -> androidx.core.content.ContextCompat
    android.support.annotation.Nullable -> androidx.annotation.Nullable
    
[Migrate instructions](https://developer.android.com/jetpack/androidx/migrate)

12343954, 2019-05-01:
you are lucky !
i don't use androidx.
i give up to fix it, waiting for the official solution.

hewking, 2019-06-03:
I want to know which version of the android.support library is used. I should not modify other third-party libraries.

hewking, 2019-06-03:
2.6.0 can use

Tino-F, 2019-06-26:
Also having the same issue

Tino-F, 2019-06-26:
Okay so I was able to resolve the issue by adding the following to android/build.gradle per https://github.com/facebook/react-native/issues/25301#issuecomment-503300537:

```
buildscript {
    ext {
        ...
        googlePlayServicesVersion = "16.+"
        firebaseVersion = "17.3.4"
    }
```

If this doesn't work for you, then you might want to look at https://github.com/facebook/react-native/issues/25301 for more details

closed 2019-09-01 by stale[bot]