react-native-camera/react-native-camera #2241: 'RNCamera' not foung in UIManager

mayankshah1607, 2019-04-27:
I'm using expo to build a react native application using this library. I'm running into an error - 'RNCamera' not found in UI Manager.

MateusAndrade, 2019-04-27:
Can you provide more info about this? Also, you followed the instructions on the README.MD about link, etc?

sibelius, 2019-04-27:
This does not work with expo directly, check docs about how to use expo camera ou link this package in your eject expo app

MateusAndrade, 2019-04-27:
> I'm using expo to build a react native application using this library. I'm running into an error - 'RNCamera' not found in UI Manager.

You can give a try at this: https://docs.expo.io/versions/latest/sdk/camera/

mayankshah1607, 2019-04-27:
Thank you for your response! Will try it and get back.

gururajphadnis, 2019-04-29:
Getting the same error message '"RNCamera" was not found in the UIManager.' in non expo environment:

"react-native": "0.57.8",
"react-native-camera": "^2.6.0",
"react-native-permissions": "^1.1.1",

jalesingh, 2019-06-18:
I getting this problem while using react-native-qrcode-scanner and now I resolved this problem after linking react-native-camera and react-native-qrcode-scanner with react-native, like below

react-native link react-native-camera
react-native link react-native-qrcode-scanner

MateusAndrade, 2019-06-22:
> Getting the same error message '"RNCamera" was not found in the UIManager.' in non expo environment:
> 
> "react-native": "0.57.8",
> "react-native-camera": "^2.6.0",
> "react-native-permissions": "^1.1.1",

You can check if react-native-camera is correctly linked to your project?  Following @jalesingh comment

ravelab, 2019-06-23:
I have the same problem on android while using react-native-qrcode-scanner but @jalesingh sollution doesn't work for me. iOS works fine

ronandowlingcullen, 2019-08-01:
Having the same problem here on android when trying to use react-native-qrcode-scanner

brkrtp, 2019-08-17:
+1 qr-code-scanner

closed 2019-10-23 by stale[bot]

robertoclavijob, 2019-11-27:
I  followed the steps to link react-native-camera **Manually**  on Android to get it working
https://react-native-community.github.io/react-native-camera/docs/installation#manual-install