react-native-camera/react-native-camera #2246: Face detection not working on android front camera

Chimaa123, 2019-05-01:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**
all version after 1.8.1

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**
Face detection is not working on Android.

**Expected Behaviour**
Face detection should work

**Steps to Reproduce**

**Additionals**
I found that this issue erased cause of #f7fb3e3 commit which is released  v1.8.1.

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-05-02:
duplicated of https://github.com/react-native-community/react-native-camera/issues/2221

Chimaa123, 2019-05-05:
#2221 is only on IOS, but this issue is on android

RWOverdijk, 2019-05-10:
@Chimaa123 What do you mean recovering? I'm having the same issue

RWOverdijk, 2019-05-10:
I missed the release. Verifying it works now.

closed 2019-05-13 by sibelius

n1ru4l, 2019-05-13:
:tada: This issue has been resolved in version 2.8.0 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera)
- [GitHub release](https://github.com/react-native-community/react-native-camera/releases/tag/v2.8.0)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

RWOverdijk, 2019-05-13:
@n1ru4l Closer, but still no. Not it's inverted horizontal causing inverted readings.