react-native-camera/react-native-camera #2248: Shows "Spinner" ; camera doesn't appear

felixchan, 2019-05-04:
- [YES ] Did you try latest release?
- [YES ] Did you try master?
- [YES ] Did you look for existing matching issues?

Deployment 8.0
React Native 0.59
Installed with the latest release (master) + `react-native link`.

When RNCamera is rendered, it shows a spinning wheel.  No camera is shown.  This occurs in both physical device and simulator.

tuicodemobile, 2019-05-04:
Have you tried the previous version?

MateusAndrade, 2019-05-04:
Are you trying this on ANdorid/IOS? Also, did you request the user for access to the camera?

felixchan, 2019-05-04:
@MateusAndrade how do I request the user for access to camera? In previous versions, it would automatically request access when <RNCamera> is rendered.

(I'm testing on iOS)

MateusAndrade, 2019-05-04:
> @MateusAndrade how do I request the user for access to camera? In previous versions, it would automatically request access when is rendered.
> 
> I also got this in XCode console (not sure if related):
> 
> ```
> 2019-05-04 12:03:08.756 [warn][tid:com.facebook.react.JavaScript] Possible Unhandled Promise Rejection (id: 0):
> TypeError: CameraManager.checkVideoAuthorizationStatus is not a function. (In 'CameraManager.checkVideoAuthorizationStatus()', 'CameraManager.checkVideoAuthorizationStatus' is undefined)
> ```
> 
> (I'm testing on iOS)

Did you link your project? IM going to give a look at the example here.

felixchan, 2019-05-04:
@MateusAndrade Yes, `react-native link react-native-camera`

MateusAndrade, 2019-05-04:
> @MateusAndrade Yes, `react-native link react-native-camera`

Hmmm, you can give a try at Cocoapods to ensure that you installed the deps properly? I only used pods whit this project. About the user request to the camera, i tested here and it should ask for the user when you render your component. I think it looks a link error on your case

musti-91, 2019-05-06:
I'm having  the same with android/IOS.
I tried to change between master and release but it's the same!

musti-91, 2019-05-06:
I solved by removing `ref` prop and passing it in child props

jqn, 2019-05-07:
@felixchan I was having the same problem. Follow the manual linking guide and make sure everything that needs to be there is where is supposed to be. In my case I was missing libRNCamera.a from my project's build phases

musti-91, 2019-05-07:
Does anyone have an issue with resolution of taken photo in IOS?
Device: IPhone 6 Plus, 7

MateusAndrade, 2019-05-07:
> Does anyone have an issue with resolution of taken photo in IOS?
> Device: IPhone 6 Plus, 7

Im using rn-camera without any resolution problem on this devices. Did you set a prop for quality or something similar to It?

musti-91, 2019-05-07:
Nope,
Configuration for save Photo: 
```javascript
const picOptions = {
      quality: 1.0,
      base64: true,
      width: 200,
      fixOrientation: true,
      pauseAfterCapture: true,
      doNotSave: false //! FOR IOS RESPONSE WILL NOT BE READ, INCLUDING (URI)
    }
```
Props: 
```javascript
captureAudio={false}
        playSoundOnCapture={true}
        autoFocus={RNCamera.Constants.AutoFocus.on}
        style={preview}
        type={cameraType}
        flashMode={RNCamera.Constants.FlashMode.auto}
```

MateusAndrade, 2019-05-07:
@musti-91 what do you mean when you say the photo is not the same?

musti-91, 2019-05-07:
@MateusAndrade not the same resolution,

MateusAndrade, 2019-05-07:
@musti-91 you can give a try without setting the width and concerting It to base64? If the problem persists you can open a New issue about that. Since this isnt the problem pointed on this issue :smile:

musti-91, 2019-05-07:
yeah, indeed.
ok, thanks

ryanyue123, 2019-05-08:
@felixchan did you check the CameraStatus for your camera. check #1744

gauthampait, 2019-05-09:
Still an issue in the latest version. Just shows the spinner. It doesn't ask for the permission as well.

felixchan, 2019-05-09:
@gauthampait I installed it manually (as well as with link) ...and now it works!

MateusAndrade, 2019-05-10:
Since the problem pointed as solved, im closing this issue. If anyone have any problem, please left a comment.

closed 2019-05-10 by MateusAndrade

switch13, 2019-06-28:
In my case, the permissions popup in iOS never appeared. It turns out I was adding it to the info.plist file that was in the root directory of the ios folder instead of adding to the info.plist located in my app's target folder.  Now I'm adding to both for good measure :)

You can find this info.plist file in xCode (I'm using Version 10.2.1) by clicking on your App project in the Project Navigator, clicking your app's target, clicking the info tab, then adding all of the appropriate Privacy Keys (ie. Privacy - Camera Usage Description) and their respective Values (Descriptions).

[image]

MaximilianoRG, 2019-09-17:
IMPORTANT:
In ios folder, executes the command "pod install" after linking manually. It solves my problem.