react-native-camera/react-native-camera #2249: XCode build error Undefined symbol: _OBJC_CLASS_$_BarcodeDetectorManagerMlkit

orhanveli, 2019-05-04:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS: 12.2
- react-native-camera: 2.6.0
- react-native:0.59.2
- react:16.8.3
- react-navigation:3.9.1

**Description/Current Behaviour**

I wanna use **Firebase ML Barcode** scanning feature in my project. I integrated `FirebaseCore` successfully but after added `Firebase/MLVision` to Profile file and `pod install`, this build error shows up

```
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$_BarcodeDetectorManagerMlkit", referenced from:
      objc-class-ref in libRNCamera.a(RNCameraManager.o)
ld: symbol(s) not found for architecture x86_64
```

here my Profile file:

```
platform :ios, '10.0'

target 'RxMPMobile' do
    
    use_frameworks!

    pod 'Firebase/Core'
    pod 'Firebase/MLVision'
    pod 'Firebase/MLVisionBarcodeModel'
end
```

**Expected Behaviour**

build success on XCode

**Steps to Reproduce**

I don't use Expo

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

orhanveli, 2019-05-17:
installing master branch solves the problem

jakecronin, 2019-05-18:
Noob question, how do you install master branch?

jakecronin, 2019-05-18:
I have the same error. Installed with: 
`npm install https://github.com/react-native-community/react-native-camera --save`
and the error persists.

Note, this library was working for me until I installed the Firebase/MLKit pods.
```
  pod 'Firebase/MLVision'
  pod 'Firebase/MLVisionTextModel'
  pod 'Firebase/MLVisionBarcodeModel'`

Now I also get this error.
```
Undefined symbols for architecture arm64:
  "_OBJC_CLASS_$_BarcodeDetectorManagerMlkit", referenced from:
      objc-class-ref in libRNCamera.a(RNCameraManager.o)
ld: symbol(s) not found for architecture arm64
`

I have been stuck on this error for a few days and would really appreciate some help!

MateusAndrade, 2019-05-18:
> Noob question, how do you install master branch?

```
yarn add react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git
```

Or

```
npm install --save react-native-camera@git+https://git@github.com/react-native-community/react-native-camera.git
```

MateusAndrade, 2019-05-18:
You can try giving a clean on the pods, and installing it again? Are opening your project on XCode with the .workspace?

MateusAndrade, 2019-05-18:
Also give a look at: https://github.com/react-native-community/react-native-camera#mostly-automatic-install-with-cocoapods. And give a look your Podfile

jakecronin, 2019-05-18:
Thanks for the suggestions @MateusAndrade, but still no luck.

Actions:
+ deleted Pods and podfile.lock 
`rm -rf Pods`
+ re-installed (pod install)
`pod install`
+ cleaned Xcode .workspace (cmd + shift + k)
+ re-build (cmd + b)

Same error appears :(

Also note, I am not using the podfile listed in that link as it drastically increased both installation time and build time; however, RNCamera and RNFirebase are both working perfectly fine as long as I do not install the Firebase/MLxxx pods.

Here is what my entire podfile:
```
# Uncomment the next line to define a global platform for your project
# platform :ios, '9.0'

target 'NativeKnifeLife' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  pod 'Firebase/Core', '~> 5.20.1'
  pod 'Firebase/Firestore', '~> 5.20.1'
  pod 'Firebase/Auth', '~> 5.20.1'
  pod 'Firebase/Messaging', '~> 5.20.1'

  #pod 'Firebase/MLVision'
  #pod 'Firebase/MLVisionTextModel'
  #pod 'Firebase/MLVisionBarcodeModel'

  #pod 'ReactNativePermissions', :path => '../node_modules/react-native-permissions'
  #pod 'react-native-camera', :path => '../node_modules/react-native-camera'
end
```

Any thoughts on what may be causing the build error? I think there may be some sort of collision between RNCamera and the Firebase/MLxxx pods, but I do not understand React Native packages intimately enough to diagnose this.

MateusAndrade, 2019-05-18:
> Thanks for the suggestions @MateusAndrade, but still no luck.

Im using the currenty podfile on a project using react-native-camera, and it works without any problem.

```ruby
# Uncomment the next line to define a global platform for your project
platform :ios, '9.0'

target 'your_app_name' do
  # Uncomment the next line if you're using Swift or would like to use dynamic frameworks
  # use_frameworks!

  # Pods for avecpay_app
  pod 'yoga', :path => '../node_modules/react-native/ReactCommon/yoga'
  pod 'React', :path => '../node_modules/react-native', :subspecs => [
    'Core',
    'CxxBridge', # Include this for RN >= 0.47
    'DevSupport', # Include this to enable In-App Devmenu if RN >= 0.43
    'RCTText',
    'RCTNetwork',
    'RCTWebSocket',
    'RCTImage',
    'RCTActionSheet',
    'RCTAnimation',
    'RCTGeolocation',
    'RCTLinkingIOS',
    'RCTSettings',
    'RCTVibration',
  ]

  # Third party deps podspec link
  pod 'DoubleConversion', :podspec => '../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec'
  pod 'glog', :podspec => '../node_modules/react-native/third-party-podspecs/glog.podspec'
  pod 'Folly', :podspec => '../node_modules/react-native/third-party-podspecs/Folly.podspec'

  pod 'react-native-camera', :path => '../node_modules/react-native-camera'

  # all of your pods form another projects 

  pod 'Firebase/Core' # using with invertase

  target 'avecpay_appTests' do
    inherit! :search_paths
    # Pods for testing
  end

  # very important to have, unless you removed React dependencies for Libraries 
  # and you rely on Cocoapods to manage it
  post_install do |installer|
    installer.pods_project.targets.each do |target|
      if target.name == "React"
        target.remove_from_project
      end
    end

    installer.pods_project.targets.each do |target|

      # The following is needed to ensure the "archive" step works in XCode.
      # It removes React & Yoga from the Pods project, as it is already included in the main project.
      # Without this, you'd see errors when you archive like:
      # "Multiple commands produce ... libReact.a"
      # "Multiple commands produce ... libyoga.a"

      targets_to_ignore = %w(React yoga)
      
      if targets_to_ignore.include? target.name
        target.remove_from_project
      end

    end

  end

end

```

When i started using react-native with pods,i ran in a lot of problems with pods, etc. You can give a try if setting the react-native pods works with you?

orhanveli, 2019-05-18:
Hello folks 

First of all installing react-native-camera with cocoapods is problematic. It installs obsolete React pod and everything gets worse. So comment react-native-camera related lines from your Profile file and execute `pod install` to get rid of them. Installing and linking manually is enough for using.

I noticed something;

react-native-camera's `react-native link` process not working correctly. I managed link process manually by adding `node_modules/react-native-camera/ios/RNCamera.xcodeproj` to `Library`in XCode resource manager. After this process something took my attention 

https://github.com/react-native-community/react-native-camera/tree/master/ios/RN

as you can see `BarcodeDetectorManagerMlkit.h` files are exists in master branch and we installed it right? But in XCode Resource manager some of these header files missing even after clean install!

**Here is my solution**

[image]

Right click **RN** folder and select `Add Files to RNCamera.xcodeproj`and navigate to `node_modules/react-native-camera/ios/RN` folder and add missing files to project. Missing files are highlighted dark already. After this process, compilation problem was gone.

I hope it helps yo too. I lost huge amount of time because of this. I'm using Firebase modules now without any problem after all. 

see ya!

MateusAndrade, 2019-05-19:
> see ya!

When you installed rn-camera with pod, did you set react-native and their dependencies on the podifle?

witalobenicio, 2019-07-09:
Anyway to solve this in a CI environment?

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius

dhavalpanchani, 2020-10-08:
In my case, I was using yarn install and I tried it with npm install(as previously I've used this) and it saved my day!

kybishop, 2021-01-05:
I was able to fix this by wiping node_modules (`rm -rf`) and then re-running `pod install`

dhavalpanchani, 2021-10-01:
> Hello folks
>
> First of all installing react-native-camera with cocoapods is problematic. It installs obsolete React pod and everything gets worse. So comment react-native-camera related lines from [...]

this worked for me, you saved my day.