react-native-camera/react-native-camera #2251: compile error com.google.android.cameraview does not exist

Deboracgs, 2019-05-04:
# Bug Report

Hello, I'm followed tutorial to installion, but when i run the command react-native run-android, i give the error.

[image]

my package.json

```
{
  "name": "ClubeTarget",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "start": "node node_modules/react-native/local-cli/cli.js start",
    "test": "jest"
  },
  "dependencies": {
    "axios": "^0.18.0",
    "lodash": "^4.17.11",
    "moment": "^2.24.0",
    "native-base": "^2.12.1",
    "react": "16.8.3",
    "react-native": "0.59.5",
    "react-native-camera": "^2.6.0",
    "react-native-cards-slider": "^0.1.2",
    "react-native-elements": "^1.1.0",
    "react-native-linear-gradient": "^2.5.4",
    "react-native-loading-spinner-overlay": "^1.0.1",
    "react-native-maps": "https://github.com/react-native-community/react-native-maps.git#c102c36d895cbef8b6a8deefd4c58fb0dc17638e",
    "react-native-permissions": "^1.1.1",
    "react-native-qrcode": "^0.2.7",
    "react-native-qrcode-scanner": "^1.1.2",
    "react-native-render-html": "^4.1.2",
    "react-native-router-flux": "^4.0.6",
    "react-redux": "^7.0.2",
    "redux": "^4.0.1",
    "redux-persist": "^5.10.0",
    "redux-saga": "^1.0.2"
  },
  "devDependencies": {
    "@babel/core": "7.4.3",
    "@babel/plugin-proposal-export-namespace-from": "^7.2.0",
    "@babel/runtime": "7.4.3",
    "babel-jest": "24.7.1",
    "jest": "24.7.1",
    "metro-react-native-babel-preset": "0.53.1",
    "react-test-renderer": "16.8.3"
  },
  "jest": {
    "preset": "react-native"
  },
  "rnpm": {
    "assets": [
      "./assets/fonts/"
    ]
  }
}
```

MateusAndrade, 2019-05-04:
Hi, can you provide more information about that? DId you run link, etc? It would be great if you provide the code repo. YOu can check our example to be sure that everything is fine on your project.

Deboracgs, 2019-05-04:
@MateusAndrade yes, i run the link, and input manually too...

In te 2.5.1 version, works fine!

MateusAndrade, 2019-05-04:
You can give a try with the master ?

closed 2019-07-11 by stale[bot]