react-native-camera/react-native-camera #2252: front camera show black screen in iOS with React Native 0.59.5

elanpang, 2019-05-05:
# Bug Report

**To Do First**
- [ ] Did you try latest release?
yes
- [ ] Did you try master?
yes
- [ ] Did you look for existing matching issues?
yes

**Platforms**
iOS

**Versions**
2.6.0

- Android:
- iOS:12.2
- react-native-camera:2.6.0
- react-native:0.59.5

**Description/Current Behaviour**
front camera show black screen with type RNCamera.Constants.Type.front
```
          <RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.face_preview}
          type={RNCamera.Constants.Type.front}
          flashMode={RNCamera.Constants.FlashMode.off}
          androidCameraPermissionOptions={{
            title: 'Permission to use camera',
            message: 'We need your permission to use your camera',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          androidRecordAudioPermissionOptions={{
            title: 'Permission to use audio recording',
            message: 'We need your permission to use your audio',
            buttonPositive: 'Ok',
            buttonNegative: 'Cancel',
          }}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
        />
```

elanpang, 2019-05-05:
resolved, just remove 
```
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.log(barcodes);
          }}
```

then the front camera works.

closed 2019-05-06 by MateusAndrade

VesperDev, 2019-06-19:
thanks so much!!!!!!!!