react-native-camera/react-native-camera #2254: Saved Photo resolution is not the same as preview

musti-91, 2019-05-08:
# Bug Report

**To Do First**

- [Yes] Did you try latest release?
- [Yes] Did you try master?
- [Yes ] Did you look for existing matching issues?

**Platforms**

Android
iOS

**Versions**

- Android: 8.0, 9.0
- iOS:12
- react-native-camera:2.6.0
- react-native: 0.59.0
- react:16.8.3
- react-navigation:3.9.1

**Description/Current Behaviour**

Saved photo resolution is different than taken photo preview.

MateusAndrade, 2019-05-08:
> # Bug Report
> **To Do First**
> 
> * [Yes] Did you try latest release?
> * [Yes] Did you try master?
> * [Yes ] Did you look for existing matching issues?
> 
> **Platforms**
> 
> Android
> iOS
> 
> **Versions**
> 
> * Android: 8.0, 9.0
> * iOS:12
> * react-native-camera:2.6.0
> * react-native: 0.59.0
> * react:16.8.3
> * react-navigation:3.9.1
> 
> **Description/Current Behaviour**
> 
> Saved photo resolution is different than taken photo preview.
> 
> **Configuration**
> 
> ```js
> const options= {
>      quality: 1.0,
>       base64: true,
>       width: 200,
>       fixOrientation: true,
>       pauseAfterCapture: true,
>       doNotSave: false 
> }
> ```
> 
> I have tried to remove the width property, but it still the same.
> If I set `skipProcessing`: true
> this leads good resolution in Android, but in IOS still the same.

You tried removing base64 or even quality?

musti-91, 2019-05-08:
I didn't change `base64`, but I changed `width: 1920` additionally to `quality:1.0` and it works.

closed 2019-05-08 by musti-91