react-native-camera/react-native-camera #2255: RNCamera crashes the app with useCamera2Api true

telmen, 2019-05-08:
# Bug Report

**To Do First**

- [x] Did you try the latest release?
- [x] Did you try the master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 6.0.1
- iOS:
- react-native-camera: 2.6.0
- react-native: 0.59.5
- react: 16.8.3
  

**Description/Current Behaviour**

When I set `useCamera2Api` prop to true, app crashes in an instant while the camera is being mounted.

**Expected Behaviour**

The app should work

**Steps to Reproduce**

**Additionals**

Here is the log of crash 
```sh
05-08 16:11:45.382 18235-18235/com.andsystems.lendpinoy.debug E/CAMERA_2::: Adjusting exposure is not currently supported for Camera2
05-08 16:11:46.213 18235-18235/com.andsystems.lendpinoy.debug E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.andsystems.lendpinoy.debug, PID: 18235
    java.lang.SecurityException: Lacking privileges to access camera service
        at android.hardware.camera2.utils.CameraBinderDecorator.throwOnError(CameraBinderDecorator.java:108)
        at android.hardware.camera2.utils.CameraBinderDecorator$CameraBinderDecoratorListener.onAfterInvocation(CameraBinderDecorator.java:73)
        at android.hardware.camera2.utils.Decorator.invoke(Decorator.java:81)
        at java.lang.reflect.Proxy.invoke(Proxy.java:393)
        at $Proxy12.createDefaultRequest(Unknown Source)
        at android.hardware.camera2.impl.CameraDeviceImpl.createCaptureRequest(CameraDeviceImpl.java:628)
        at com.google.android.cameraview.Camera2.startCaptureSession(Camera2.java:802)
        at com.google.android.cameraview.Camera2$1.onOpened(Camera2.java:93)
        at android.hardware.camera2.impl.CameraDeviceImpl$1.run(CameraDeviceImpl.java:134)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:148)
        at android.app.ActivityThread.main(ActivityThread.java:5417)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

```
> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-05-08:
You setting any adjust on the picture? Looking to the following lime: `05-08 16:11:45.382 18235-18235/com.andsystems.lendpinoy.debug E/CAMERA_2::: Adjusting exposure is not currently supported for Camera2` .

Algo did you asked the user or the permission to access the camera, and this have been granted? Im going to give a look about camera2api on the project. If its possible, i think you can use rn-camera without any problem without setting this prop

telmen, 2019-05-09:
> You setting any adjust on the picture? Looking to the following lime: `05-08 16:11:45.382 18235-18235/com.andsystems.lendpinoy.debug E/CAMERA_2::: Adjusting exposure is not currently supported for Camera2` .

I think it's not relevant to taking a picture. 

> Algo did you asked the user or the permission to access the camera, and this have been granted? Im going to give a look about camera2api on the project. If its possible, i think you can use rn-camera without any problem without setting this prop

The app crashes when camera mounts and I checked camera permission it's already granted.

telmen, 2019-05-09:
> If its possible, i think you can use rn-camera without any problem without setting this prop

The reason I use this prop is in Camera1 API there's a bug with orientation I think(_front camera face detection is not working and in order to be it working I had to rotate the device 90 deg_). I'm trying to check bug still happens in Camera2.

MateusAndrade, 2019-05-09:
There are some problems related to face recognition with orientation, etc. 

Im trying to test the cenarios that mah cause this problem, If you provide info about the tests you made, It Will help a lot.

closed 2019-07-15 by stale[bot]