react-native-camera/react-native-camera #2258: Release build fails when ProGuard is enabled

andresmtz98, 2019-05-09:
# Bug Report

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9.0
- react-native-camera: 2.6.0 | 2.7.0
- react-native: 0.59.5
- react: 16.8.3

**Description/Current Behaviour**

When `def enableProguardInReleaseBuilds = true` the release build fails.

**Expected Behaviour**

Build succeded.

**Steps to Reproduce**
- Create any blank project
- Set up this library for Android and the release config
- Run `./gradlew assembleRelease` or `./gradlew bundleRelease`

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2019-07-15 by stale[bot]

andresmtz98, 2019-07-15:
@sibelius Bro, can you please reopen this issue?

andresmtz98, 2019-07-15:
@sibelius Is there any plan to adress this issue?

mayconmesquita, 2020-01-06:
I'm getting this problem here