react-native-camera/react-native-camera #2259: Mouth open face

RWOverdijk, 2019-05-10:
# Feature-Request

**Describe the Feature**

It would be nice to be able to detect whether or not the user's mouth is open.

**Possible Implementations**

I have implemented this to some degree (pun intended) of success with the following:

```
const { rightMouthPosition, leftMouthPosition, bottomMouthPosition } = face;
const { x: x1, y: y1 } = leftMouthPosition;
const { x: x2, y: y2 } = rightMouthPosition;
const { x: x3, y: y3 } = bottomMouthPosition;
const y = Math.sqrt(Math.pow(x2 - x1, 2) + Math.pow(y2 - y1, 2));
const x = Math.sqrt(Math.pow(x3 - x2, 2) + Math.pow(y3 - y2, 2));
const z = Math.sqrt(Math.pow(x3 - x1, 2) + Math.pow(y3 - y1, 2));

const deg = Math.acos((Math.pow(z, 2) + Math.pow(x, 2) - Math.pow(y, 2)) / (2 * z * x)) * (180 / Math.PI);

if (deg < 120) {
  // Mouth open
}
```

I calculate a triangle for the mouth's edges and check the angle of the bottom corner (bottom of mouth). This works quite well (probability can be calculated by using ranges).

The problem here is with the roll and yaw angles. These would have to somehow have to change the coordinates for the calculation which is a bit beyond my math knowledge I'm afraid.

sibelius, 2019-05-10:
can this be done on javascript only?

RWOverdijk, 2019-05-10:
I'm pretty sure it can. It's a guesstimate of course, but so are the other values. If anyone can help with the transform calculation that is... It's the space where X and Y both have been applied.

closed 2019-07-16 by stale[bot]

GH974, 2019-09-14:
Hi, someone to find a solution?