react-native-camera/react-native-camera #2261: Android front facing camera performance

RWOverdijk, 2019-05-12:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: Any
- iOS: N/A
- react-native-camera: Master/2.7.0	
- react-native: 0.59.5
- react: 16.8.3

**Description/Current Behaviour**

Using the example from the repository, when flipping to the front facing camera and selecting "detect faces" the resulting "matches" box is rotated -90 degrees and detection is all over the place, resulting in very weird readings.

[image]

**Expected Behaviour**

Same as iOS.

**Steps to Reproduce**

Using the example from the repository, when flipping to the front facing camera and selecting "detect faces".

**Additionals**

Additionally, the readings seem inverted, too. iOS works perfectly, android is just very unstable.

RWOverdijk, 2019-05-12:
**Update:** I can confirm that adding back the rotation makes the results flip back upright but the readings are still very unstable.

https://github.com/react-native-community/react-native-camera/commit/f7fb3e361d4f1b773ee93bea8336f276e84b692c#diff-8da71ab3d0d99d3b176b2261137b0707R280

andreseiji, 2019-05-13:
I am facing the same issue here. @RWOverdijk, how did you get the rotation up?

About the fix, before the https://github.com/react-native-community/react-native-camera/pull/2257 merge, the front camera (on Android) hardly recognised any face. I have tested it on 4 different devices and it is not perfect, but far better than before.

However, the face detection is still rotated 90 degrees counter clockwise.

RWOverdijk, 2019-05-13:
@andreseiji Correct, when rotated back it recognises my hair as a bunch of faces, or nothing at all. Something funky is going on and I couldn't find it in the time I spent looking for it.

andreseiji, 2019-05-13:
@RWOverdijk Just opened a PR which should fix the rotation issue :)

RWOverdijk, 2019-05-13:
@andreseiji Yeah I saw! I commented on the other issue. 

> @n1ru4l Closer, but still no. Not it's inverted horizontal causing inverted readings.

closed 2019-07-24 by stale[bot]