react-native-camera/react-native-camera #2267: ReactNativeJS: TypeError: undefined is not an object (evaluating '_reactNativeCamera.default.constants')

liuyulin666, 2019-05-15:
"react-native-camera": "^2.1.1",
"react-native": "^0.59.3",

Online version is right, the branch error, on the premise of not do version update, could you tell me how to solve this？？？River's lake emergency, thank you！！！

liuyulin666, 2019-05-15:
This error is located at:
        in Index (created by SceneView)
        in SceneView (at StackViewLayout.js:786)
        in RCTView (at View.js:45)
        in View (at StackViewLayout.js:785)
        in RCTView (at View.js:45)
        in View (at StackViewLayout.js:784)
        in RCTView (at View.js:45)
        in View (at createAnimatedComponent.js:151)
        in AnimatedComponent (at StackViewCard.js:69)
        in RCTView (at View.js:45)
        in View (at createAnimatedComponent.js:151)
        in AnimatedComponent (at screens.native.js:59)
        in Screen (at StackViewCard.js:57)
        in Card (at createPointerEventsContainer.js:27)
        in Container (at StackViewLayout.js:862)
        in RCTView (at View.js:45)
        in View (at screens.native.js:83)
        in ScreenContainer (at StackViewLayout.js:313)
        in RCTView (at View.js:45)
        in View (at createAnimatedComponent.js:151)
        in AnimatedComponent (at StackViewLayout.js:309)
        in Handler (at StackViewLayout.js:302)
        in StackViewLayout (at withOrientation.js:30)
        in withOrientation (at StackView.js:79)
        in RCTView (at View.js:45)
        in View (at Transitioner.js:214)
        in Transitioner (at StackView.js:22)
        in StackView (created by Navigator)
        in Navigator (at createKeyboardAwareNavigator.js:12)
        in KeyboardAwareNavigator (at createAppContainer.js:387)
        in NavigationContainer (at App.js:160)
        in App (at renderApplication.js:35)
        in RCTView (at View.js:45)
        in View (at AppContainer.js:98)
        in RCTView (at View.js:45)
        in View (at react-native-root-siblings/index.js:33)
        in RootSiblingsWrapper (at AppContainer.js:112)
        in RCTView (at View.js:45)
        in View (at AppContainer.js:115)
        in AppContainer (at renderApplication.js:34)
    
    This error is located at:
        in NavigationContainer (at App.js:160)
        in App (at renderApplication.js:35)
        in RCTView (at View.js:45)
        in View (at AppContainer.js:98)
        in RCTView (at View.js:45)
        in View (at react-native-root-siblings/index.js:33)
        in RootSiblingsWrapper (at AppContainer.js:112)
        in RCTView (at View.js:45)
        in View (at AppContainer.js:115)
        in AppContainer (at renderApplication.js:34)

Stat1c14, 2019-05-17:
Not sure if this will help you, but I was receiving a very similar error caused by doing

`import RNCamera from 'react-native-camera';`

instead of 

`import { RNCamera } from 'react-native-camera';`

Hopefully that helps.

MateusAndrade, 2019-05-18:
Are you importing RNCamera doing: 

```javascript
import { RNCamera, FaceDetector } from 'react-native-camera';
```

As @Stat1c14 said, you need to apply a destructing on your import, or use something like:
```javascript
   import RNCameraImport from 'react-native-camera';
```

And access it with ```RNCameraImport.RNCamera()```. Btw, i recommend using the first option, since its more clear and easy to understand.

Also, you can check if you are accessing the RNCamera constants with ```RNCamera.Constants```?

Petrando, 2019-07-22:
I am experiencing the same issue with RN version 0.95.5, after npm install and link the RN camera library and then RN start --reset-cache. I'm trying to create a RN barcode scanner by [this tutorial](https://medium.com/@dinukadilshanfernando/implementing-a-barcode-scanner-by-using-react-native-camera-b170de4b7f51) . At first i use 

`import Camera from 'react-native-camera'`

 as suggested by said tutorial, and got error like below 

[image]

which seems similiar to [this stackoverflow question](https://stackoverflow.com/questions/55982666/how-to-fix-typeerror-undefined-is-not-an-object-evaluating-reactnativecamer).

Then as suggested by @MateusAndrade above, i change my import into 

`import RNCameraImport from 'react-native-camera';`
`const RNCamera = RNCameraImport.RNCamera();`

And change the Camera component invocation into this :

 `<RNCamera
			style={styles.preview}
			torchMode={this.state.torchOn ? RNCamera.Constants.TorchMode.on : RNCamera.Constants.TorchMode.off}
			onBarCodeRead={this.onBarCodeRead}
			ref={cam => this.camera = cam}
			aspect={RNCamera.Constants.Aspect.fill}
		  ></RNCamera>`
(Notice above that i used capital starting letter at `RNCamera.Constants` call, as suggested by @MateusAndrade ) 
which then i got this error below : 

[image]

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius

Luis-Kendelick, 2019-09-16:
> I am experiencing the same issue with RN version 0.95.5, after npm install and link the RN camera library and then RN start --reset-cache. I'm trying to create a RN barcode scanner by [this tutorial](https://medium.com/@dinukadilshanfernando/implementing-a-barcode-scanner-by-using-react-native-camera-b170de4b7f51) . At first i use
> 
> `import Camera from 'react-native-camera'`
> 
> as suggested by said tutorial, and got error like below
> 
> [image]
> 
> which seems similiar to [this stackoverflow question](https://stackoverflow.com/questions/55982666/how-to-fix-typeerror-undefined-is-not-an-object-evaluating-reactnativecamer).
> 
> Then as suggested by @MateusAndrade above, i change my import into
> 
> `import RNCameraImport from 'react-native-camera';`
> `const RNCamera = RNCameraImport.RNCamera();`
> 
> And change the Camera component invocation like so :
> 
> `<RNCamera style={styles.preview} torchMode={this.state.torchOn ? RNCamera.Constants.TorchMode.on : RNCamera.Constants.TorchMode.off} onBarCodeRead={this.onBarCodeRead} ref={cam => this.camera = cam} aspect={RNCamera.Constants.Aspect.fill} ></RNCamera>`
> (Notice above that i used capital starting letter at `RNCamera.Constants` call )
> then i got this error below :
> 
> [image]

man, you're using **TorchMode.on** and **TorchMode.off**?
If yes, they updated the package. You have to use FlashMode.torch and FlashMode.off instead.

Luis-Kendelick, 2019-09-16:
guys, "constants" must be "Constants". 
TorchMode.on and TorchMode.off no more exists. There are FlashMode.torch and FlashMode.off in the documentation.
They updated the package.

nihp, 2019-12-05:
I am also getting the same error. Any solution for this?

aqeebimtiaz, 2020-01-16:
I'm getting this same error still, any solutions yet?

> I am experiencing the same issue with RN version 0.95.5, after npm install and link the RN camera library and then RN start --reset-cache. I'm trying to create a RN barcode scanner by [this [...]

rahultectum, 2020-05-02:
[image]

I am also getting the same error. Any solution for this?

YashKumarVerma, 2020-06-12:
Nothing mentioned above works. :cry:

renatobenks, 2020-06-12:
which version are you using @YashKumarVerma?

ayanmohsin, 2022-07-15:
Please How you resolved kindly guide me

Aakash359, 2022-09-12:
Hello I am using React-native-camera with typescript project but in my case its giving error like tyepError : RN Camera can not be undefined , can anyone let me know what issues is there I am just using example code of react-native-camera
[image]

ayanmohsin, 2022-09-12:
Dear React Native CAmera is Depresable in the favour of react native vision
camera so use react native vision camera library in react native> 6.0