react-native-camera/react-native-camera #2268: How to turn off the record audio permission request

000xuandu, 2019-05-15:
# I'm building an application that scans the qr code with the react-native-camera (RNCamera) library.
Currently, I don't have a recording-related function, so I don't want to ask my users to accept "record audio" permissions.
I have read the material in a different way, but failed.
Please help me.

**I do not want to appear dialog box asking for audio recording rights as shown in Figure 2**

[
[image]
[image]

](url)

MateusAndrade, 2019-05-15:
You can pass: `captureAudio={false}` to the RNCamera component.

MateusAndrade, 2019-05-15:
> You can pass: `captureAudio={false}` to the RNCamera component.

```jsx
<RNCamera
          ref={ref => {
            this.camera = ref;
          }}
          style={styles.preview}
          type={RNCamera.Constants.Type.back}
          flashMode={RNCamera.Constants.FlashMode.on}
          captureAudio={false}
          onGoogleVisionBarcodesDetected={({ barcodes }) => {
            console.warn(barcodes[0].data);
            if (!Utils.isNull(barcodes[0].data) & this.isCan) {
              this.runAction(barcodes[0].data)
            }
          }}
/>
```

closed 2019-05-15 by MateusAndrade