react-native-camera/react-native-camera #2274: Last Stable version??

karanpratapsingh, 2019-05-16:
can someone tell me the last stable version??

I just need a simple camera view with qr code scanning capability, can't seem to get v2.9.0 work on android

Thanks

karanpratapsingh, 2019-05-16:
for anyone wondering use v1.13.1 it's the most stable one if you just need this module for simple stuff like barcode scanning

andresmtz98, 2019-05-17:
> for anyone wondering use v1.13.1 it's the most stable one if you just need this module for simple stuff like barcode scanning

Bro do you have a guide to set up that version?

MateusAndrade, 2019-05-18:
> > for anyone wondering use v1.13.1 it's the most stable one if you just need this module for simple stuff like barcode scanning
> 
> Bro do you have a guide to set up that version?

```
yarn add react-native-camera@1.13.1
```

You can set the version you want, just changing the @1.13.1. After doing this, probably you will need to do a pod install, etc.

closed 2019-07-24 by stale[bot]