react-native-camera/react-native-camera #2275: CameraManager.checkVideoAuthorizationStatus is not a function

orhanveli, 2019-05-16:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Suddenly I started to get this warning and camera view doesn't work. 
I uninstall install an link again, nothing changed. Am I missng something?

```
Possible Unhandled Promise Rejection (id: 0):
TypeError: CameraManager.checkVideoAuthorizationStatus is not a function
TypeError: CameraManager.checkVideoAuthorizationStatus is not a function
    at requestPermissions$ (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:163060:61)
    at tryCatch (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:25862:19)
    at Generator.invoke [as _invoke] (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:26037:24)
    at Generator.prototype.<computed> [as next] (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:25905:23)
    at tryCatch (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:25862:19)
    at invoke (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:25938:22)
    at blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:25968:13
    at tryCallTwo (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:3165:7)
    at doResolve (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:3329:15)
    at new Promise (blob:http://localhost:8081/0eaaec84-bf77-4bb0-9d79-9a76b9ab6188:3188:5)
```

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

orhanveli, 2019-05-17:
Hello everyone I found the problem which is `react-native link react-native-camera` doesn't work as expected. Manually linking library solved the problem

closed 2019-05-17 by orhanveli

MaximilianoRG, 2019-09-17:
IMPORTANT:
In ios folder , execute the command "pod install" after linking manually the library

PinelliaTime, 2019-10-14:
Hello,i'm trying you method,but also have the same problems,is there another methods to solve this  problem? Hopeing for your methods,thanks very much!
`"react-native": "0.60.5",
"react-native-camera": "^3.7.1",`

faridrehila, 2020-01-04:
I also had the same issue and solved it by adding this line in my Podfile : 
`pod 'react-native-camera', path: '../node_modules/react-native-camera'`

The linking probably didn't work properly when installing the library. I also tried installing the library manually but it also didn't work properly. 

`"react-native": "0.60.5", "react-native-camera": "^3.7.1"`

fedor-moiseev, 2021-05-24:
I have the same error with Expo