react-native-camera/react-native-camera #2277: Feat. Request: use ML Kit face detector for single file

andreseiji, 2019-05-17:
# Feature-Request

**Describe the Feature**
A method where the app send an image captured with React Native Camera, and Firebase ML Kit analysis it returning an array of faces for that image.

**Possible Implementations**
For Android, there is already a class named `FileFaceDetectionAsyncTask` which already uses Firebase ML Kit for face detection. What I could not find is a method to send an specific file, instead of the camera frames.

**Related Issues**
-

andreseiji, 2019-05-17:
**Update:** There is already a method which does exactly what I proposed, found it searching the source code. It is just really slow. I will try to figure why, and how to make it faster, since the live one is much faster.

For future reference, I suggest an update to the https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md under `Face Detection Related props`.

```
### Face Detection for Single File 

A single URI can be sent for Face Detection using `detectFacesAsync` method. The method has two params:

`uri`: The image URI
`options`: `mode`, `detectLandmarks`,  `runClassifications`

Usage:

```
import { FaceDetector} from 'react-native-camera';
...
const img = await FaceDetector.detectFacesAsync(data.uri);
```
```

andreseiji, 2019-05-20:
After some testing on the `.detectFacesAsync` method, it works, but it takes a very long time (about 15s on a Galaxy S5). So, I am reopening this.

I believe the long time may be caused by writing and reading of the image file, since now it's an URI. Maybe the RN Core affects the time too, as stated here https://github.com/react-native-community/react-native-camera/issues/1836#issuecomment-463784038.

I do not have much native experience, but I am trying to solve this. What I propose is an option to run the `FirebaseVisionFaceDetector` on the taken picture, and return the picture data alongside the ML Kit data for face detection. Hopefully, processing this in the native layer will make things way faster.

I know it is a very specific case and, honestly, it is a case I need to solve for a project. I will keep on trying to make this work, and surely would appreciate any thoughts the community's got on this.

closed 2019-07-26 by stale[bot]

TrueTiem, 2020-09-27:
@andreseiji I know it's been a while but detectFacesAsync method still takes a very long time. Were you able to find and fix the issue or have any possible cause to this?

andreseiji, 2020-09-29:
@TrueTiem Back then, I could not find a good solution to improve the `.detectFacesAsync` method. I believe in parts, the time is due to image I/O and the React Native bridge. The live method works well and fast, since a frame is smaller than a taken picture.

The issue itself has never been fixed. What I did was using the last frame state of my live capture, which was fast enough for my needs.

JayasimhaDev, 2023-05-28:
I am also using face detection in Image uri but [image]

 Im getting this error how to solve this one