react-native-camera/react-native-camera #2279: Interleaved2of5 with only 4 digits not recognized

hoffmana, 2019-05-20:
Hi,
we are having Interleaved2of5 codes with 4 and 6 digits like 1234 and 345678:
[image]
[image]

Our scanner is having no problem to detect 6 digits code, we get the result after scanning:
{type: "ITF", data: "042826"}

But if the code contains only 4 digits, the scanner does not detect code at all.

Is anyone facing same or has some hints, why there is an issue detecting 4 dgiti code? Are there any restrictions for the scanner/barcode?

I appreciate your help, thanks a lot.

Best regards,

Thomas

MateusAndrade, 2019-05-20:
@hoffmana you can edit your comment with the issue template?

hoffmana, 2019-05-20:
Sorry,. I am quite new to GitHub. What do you mean exactly? I added the information, that it happens with Android. iOS is detecting correctly.

MateusAndrade, 2019-05-26:
You can test the barcode with this app: https://play.google.com/store/apps/details?id=com.google.zxing.client.android,  and tell me if you got the same problem?

hoffmana, 2019-05-26:
Hi, I just tested this app, but it also does not recognized the 4 digit code, but a 6 digit code.
[image]
[image]
If you scan it with SCANDIT (https://play.google.com/store/apps/details?id=com.scandit.demoapp), then it is recognized perfectly.
[image]

MateusAndrade, 2019-05-26:
I think this is a problem with zxing, the Android library used to Scan barcodes. You tried with ML Kit?

closed 2019-08-01 by stale[bot]