react-native-camera/react-native-camera #2285: iOS - yellow warning "Class RNFaceDetectorModuleMLKit was not exported. Did you forget to use RCT_EXPORT_MODULE()?"

truongluong1314520, 2019-05-27:
# Bug Report

**Platforms**
IOS

**Versions**

- Android:
- iOS: 12.2
- react-native-camera: 2.9.0
- react-native: 0.57.0
- react: 16.5.0
- react-navigation: 2.14.2

**Description/Current Behaviour**

When I run the app, there is a warning in the first place: "Class RNFaceDetectorModuleMLKit was not exported. Did you forget to use RCT_EXPORT_MODULE()?"

The app only uses the camera for taking image, so I only follow your instruction of [iOS tab](https://github.com/react-native-community/react-native-camera#ios) no [face/text/barcode detection](https://github.com/react-native-community/react-native-camera#ios)

**Expected Behaviour**

There shouldn't be a warning

**Steps to Reproduce**

Install manually follow the [iOS tab](https://github.com/react-native-community/react-native-camera#ios) 

**Additionals**

[image]

**Environment**

React Native Environment Info:
    System:
      OS: macOS 10.14.5
      CPU: x64 Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz
      Memory: 28.10 MB / 16.00 GB
      Shell: 3.2.57 - /bin/bash
    Binaries:
      Node: 10.15.0 - /usr/local/bin/node
      Yarn: 1.13.0 - /usr/local/bin/yarn
      npm: 6.9.0 - /usr/local/bin/npm
      Watchman: 4.9.0 - /usr/local/bin/watchman
    SDKs:
      iOS SDK:
        Platforms: iOS 12.2, macOS 10.14, tvOS 12.2, watchOS 5.2
      Android SDK:
        Build Tools: 23.0.1, 25.0.2, 26.0.1, 26.0.2, 26.0.3, 27.0.3, 28.0.2, 28.0.3
        API Levels: 16, 17, 18, 19, 21, 22, 23, 24, 25, 26, 27, 28
    IDEs:
      Android Studio: 3.1 AI-173.4907809
      Xcode: 10.2.1/10E1001 - /usr/bin/xcodebuild
    npmPackages:
      react: 16.5.0 => 16.5.0 
      react-native: 0.57.0 => 0.57.0 
    npmGlobalPackages:
      react-native-cli: 2.0.1

Thank you in advance!

sibelius, 2019-05-27:
it looks like your link is wrong

MateusAndrade, 2019-05-27:
Which version are you using?

truongluong1314520, 2019-05-28:
Hi @sibelius, I will try to link it again

Hi @MateusAndrade, I am using:
- react-native-camera: 2.9.0
- react-native: 0.57.0

truongluong1314520, 2019-05-28:
Hi @sibelius , I follow exactly the instruction: 

- npm install react-native-camera --save
- In XCode, in the project navigator, right click Libraries ➜ Add Files to [your project's name]
- Go to node_modules ➜ react-native-camera and add RNCamera.xcodeproj
- Expand the RNCamera.xcodeproj ➜ Products folder
- In XCode, in the project navigator, select your project. Add libRNCamera.a to your project's Build Phases ➜ Link Binary With Libraries
- Click RNCamera.xcodeproj in the project navigator and go the Build Settings tab. Make sure 'All' is toggled on (instead of 'Basic'). In the Search Paths section, look for Header Search Paths and make sure it contains both $(SRCROOT)/../../react-native/React and $(SRCROOT)/../../../React - mark both as recursive.

However, the warning is still there.

[image]

[image]

[image]

aidanw, 2019-05-28:
I have the same issue. 

"react-native": "0.57.7",
"react-native-camera": "^2.8.0"

truongluong1314520, 2019-05-29:
Hi @aidanw , did you link the library manually, automatically, or using Pod? In my case, I link manually

pgzm29, 2019-05-29:
Hi @truongluong1314520 I have the exact same issue, same versions and everything. I linked the library manually. 
I tried the Pod install as well but it messed everything up. I reinstalled the library and relinked manually and the issue persists. Have you had any luck with this?

truongluong1314520, 2019-05-30:
Hi @pgzm29 , I am not reinstalled the library since the we don't change anything inside the node_modules. However, relinked manually doesn't resolve the issue

MateusAndrade, 2019-05-30:
@truongluong1314520 you can provide the code that you are using to call rn-camera? :)

pgzm29, 2019-05-31:
> @truongluong1314520 you can provide the code that you are using to call rn-camera? :)

Hi MateusAndrade, I think I'm doing the same thing as @truongluong1314520 ... the warning appears as soon as you launch the app after you link the library to your project. Even before adding the code to call rn-camera. But in any case, the way I'm calling rn-camera is the following: 

`import { RNCamera } from 'react-native-camera';`

Is that what you're referring to? :)

MateusAndrade, 2019-05-31:
> Is that what you're referring to? :)

I mean how you are calling on the render. Sorry about the confusion. You can show me how you render rn-camera similar to this:

```jsx
<RNCamera {...props}>
{children}
</RNCamera>
```

pgzm29, 2019-05-31:
Yes, exactly like that. And it seems it's working correctly, but the issue that started this thread is still showing. I don't know if this can cause any problem later when the app is in production or in any other case.

MateusAndrade, 2019-05-31:
@pgzm29 you please test if this PR solves your problem: https://github.com/MateusAndrade/react-native-camera/tree/bugfix/rnface-detect-ext-export.

Just change you package.json to: 

```json
    "react-native-camera": "MateusAndrade/react-native-camera#bugfix/rnface-detect-ext-export",
```

I did a little change on `RNFaceDetectorModuleMLKit.m`. If this solves your problem, i can send a PR with this. :)

pgzm29, 2019-05-31:
Thanks for working on this so quickly, unfortunately, the issue still persists :/

truongluong1314520, 2019-06-01:
Hi @MateusAndrade , I am really appreciated your helping. Below if the code I am using for rn-camera:

    <RNCamera
        ref={ref => {
            this._camera = ref;
        }}
        style={styles.preview}
        type={RNCamera.Constants.Type.back}
        flashMode={RNCamera.Constants.FlashMode.auto}
        captureAudio={false}
    />

I also try to do what you said but warning is still there

Hope the code of <RNCamera/> can help you somehow

Thank you

Isaacs9, 2019-06-05:
I'm having the same issues with the same versions.

sibelius, 2019-06-05:
Avoid manual linking, use cocoapods

aweffr, 2019-06-06:
> @pgzm29 you please test if this PR solves your problem: https://github.com/MateusAndrade/react-native-camera/tree/bugfix/rnface-detect-ext-export.
>
> Just change you package.json to:
>
> ```json [...]

sorry, I tried to change that file but the problem doesn't resolve. I just start a new 0.57.8 project and install react-native-camera, following the cocapods config

truongluong1314520, 2019-06-17:
Hi @sibelius , it looks like cocoapods is also having the issue, please take a look for us. Thank you in advance!

sibelius, 2019-08-01:
use autolinking with v3

closed 2019-08-01 by sibelius

inkCrazy, 2019-10-12:
I have the same problem!
"react-native-camera": "^2.11.1",
"react-native": "0.57.4",