react-native-camera/react-native-camera #2294: ERROR: Failed to resolve: support-v4

1512751361, 2019-06-03:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:2.10.0
- react-native:0.59.8
- react:16.8.3
  

**Description/Current Behaviour**
打开android stuido报下面这个错误：
ERROR: Failed to resolve: support-v4 
Affected Modules: react-native-camera
如果使用react-native-camera@2.9.0版本不会报这个错误

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

ouabing, 2019-06-03:
The 2.10.0 version seems to migrate to AndroidX, just lock to 2.9.0 if your dependencies are not fully migrated to AndroidX. Or you can follow [this](https://developer.android.com/jetpack/androidx/migrate) to make your project support AndroidX.

I think it will be better for RNCamera to start a new version like 3.0.0 with AndroidX support, and release 2.x without AndroidX support. This will allow people who is not ready to migrate to AndroidX to access the latest bug fixes and features.

2.10.0 迁移到 AndroidX 上了，RN 0.60.0 之前先用 2.9.0 吧，很多依赖库都还没迁移到 AndroidX，如果你确定依赖库都迁移了可以[看这个](https://developer.android.com/jetpack/androidx/migrate) 迁移到 AndroidX .

closed 2019-06-03 by 1512751361

n1ru4l, 2019-06-06:
@ouabing @1512751361 I reverted the breaking change. You should be able to safely upgrade to 2.10.1 

Progress for AndroidX is now tracked in #2306

Sorry for causing the inconveniences!