react-native-camera/react-native-camera #2296: improve AndroidX migration

sibelius, 2019-06-04:
follow this https://github.com/react-native-community/react-native-camera/pull/2068#issuecomment-498781811

xaphod, 2019-06-06:
I can't get from 2.9.0 -> 2.10.0 because of this issue (my project doesn't use AndroidX, and cannot migrate to it; 2.10.0 forces it to happen which causes my build to break).
I have Android orientation issues on 2.9.0, which may be fixed in 2.10.0 -- no idea, I cannot run it, but I hope.
So I hope this can get some love soon. I'd help if I was able (I don't know android / RN well)

n1ru4l, 2019-06-06:
Related #2302

@sibelius We should not ship breaking changes in a minor version. IMHO we should revert #2068 and do a new fix version.

After we have a correct migration path we can ship Android X support.

brunobar79, 2019-06-18:
Why not having a major version bump + a note in the readme for react-native < 60 versions to use the older version ?

n1ru4l, 2019-06-18:
Because there is no documented migration path and users will be confused. react-native 0.60 is still in the release candidate stage.

brunobar79, 2019-06-18:
I'm confused about the "migration path". Isn't it a breaking change? That should be acceptable as long as you do a major version bump.  And the conversation is happening here: https://github.com/react-native-community/discussions-and-proposals/issues/129

Also, IMO would be nice not to block people to upgrade to 0.60 once it's out, maybe have a branch ready and then release at the same time.

mikehardy, 2019-06-18:
androidx is breaking, but it's possible to support legacy support API consumers - you integrate the "bob" package which the community has patched to create a "reverse-jetified aar" for libraries, that legacy API consumers may use. Just some package.json lines and a bit of docs and you publish that aar at the same time when you release

mikehardy, 2019-06-18:
and for people that use androidx now, I tested react-native-camera after converting my app to AndroidX and using `npm i jetifier && npx jetify` and it works also. So there is forward and backward compatibility in places, it's just all really brand new so it isn't packaged well, tested well, or documented yet. But success or failure reports for bob or jetify packages are welcome. Cheers

sibelius, 2019-07-12:
comment in this pull request if everything is working

https://github.com/react-native-community/react-native-camera/pull/2306

then we can do a new release

closed 2019-07-12 by sibelius