react-native-camera/react-native-camera #2297: Add autoconfig when linking or add procedural configuration for issue #2063 missingDimensionStrategy

skhameneh, 2019-06-04:
# Feature-Request

**Describe the Feature**
Better zeroconfig / autoconfig / autolinking, resolve manual `missingDimensionStrategy`

**Possible Implementations**
* Autoconfig upon linking.
* Add to `yo rn-toolbox` see https://github.com/bamlab/generator-rn-toolbox

**Related Issues**
https://github.com/react-native-community/react-native-camera/issues/2063

> Support react-native-camera! 👉  https://opencollective.com/react-native-camera/donate
> Want this resolved faster? Add a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-06-04:
check autolinking here https://github.com/react-native-community/cli/blob/master/docs/autolinking.md

skhameneh, 2019-06-05:
@sibelius autolinking doesn't configure `missingDimensionStrategy`.

sibelius, 2019-06-05:
you can use a hook to add `missingDimesionStrategy`, check here https://github.com/react-native-community/cli/blob/master/docs/dependencies.md#hooks

skhameneh, 2019-08-04:
@sibelius that's great, but it's a far stretch to call that config driven and it's definitely not zeroconfig.  That's not a solution.