react-native-camera/react-native-camera #2302: Android Build fails

victorkvarghese, 2019-06-06:
# Bug Report

**To Do First**

- [ YES ] Did you try latest release?
- [ YES ] Did you try master?
- [ YES] Did you look for existing matching issues?

**Platforms**
Android

**Versions** Latest one. tried from 2.4.0

[image]

I tried various solutions from the issues .. did not help
Basic install and linking using react-native-link

n1ru4l, 2019-06-06:
It seems like the project is not properly linked. Please follow the manual link instructions:

https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#android

JuAlexandre, 2019-06-06:
I'm facing the same problem as I followed the documentation literally

JuAlexandre, 2019-06-06:
After several hours of testing I have this error message while I followed the line-by-line

`ERROR: Manifest merger failed : Attribute application@appComponentFactory value=(androidx.core.app.CoreComponentFactory) from [androidx.core:core:1.0.0] AndroidManifest.xml:22:18-86
	is also present at [com.android.support:support-compat:28.0.0] AndroidManifest.xml:22:18-91 value=(android.support.v4.app.CoreComponentFactory).
	Suggestion: add 'tools:replace="android:appComponentFactory"' to <application> element at AndroidManifest.xml:7:5-117 to override.
`

n1ru4l, 2019-06-06:
Might be related to #2068

Could you please test if you do still experience these issues with a clean `react-native-camera@2.9.0` installation?

`yarn add -E react-native-camera@2.9.0`

or

`npm install -E react-native-camera@2.9.0`

Victor-Varghese, 2019-06-06:
I just updated Android Studio to latest and I was able to bypass the first errors. But stuck at @JuAlexandre 's error

```
ERROR: Manifest merger failed : Attribute application@appComponentFactory value=(androidx.core.app.CoreComponentFactory) from [androidx.core:core:1.0.0] AndroidManifest.xml:22:18-86
	is also present at [com.android.support:support-compat:28.0.0] AndroidManifest.xml:22:18-91 value=(android.support.v4.app.CoreComponentFactory).
	Suggestion: add 'tools:replace="android:appComponentFactory"' to <application> element at AndroidManifest.xml:7:5-117 to override.
```

Victor-Varghese, 2019-06-06:
@n1ru4l  Updating android studio to latest and using 2.9.0 worked for me

n1ru4l, 2019-06-06:
@Victor-Varghese I am sorry that you had to experience this pain. I reverted the commit that breaked the backwards compat in version https://github.com/react-native-community/react-native-camera/releases/tag/v2.10.1

The AndroidX support will be reintroduced in a major release once there is a clear and documented migration path. 

For now, I am going to close this issue and I suggest you upgrading to v2.10.1.

closed 2019-06-06 by n1ru4l