react-native-camera/react-native-camera #2309: After focus on single tap, calling takePicture cause focus changes and result is blurry

winadiw, 2019-06-10:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 9.0, Samsung Galaxy A70
- iOS:
- react-native-camera: 2.10.1
- react-native: 0.57.0
- react: 16.5.0
  

**Description/Current Behaviour**

- Auto focus set to off
- Touch on camera preview, the preview focus on the image correctly
- Press a button to call takePicture
- The camera preview focus off (reverting to previous out of focus before tapping the preview) 
- The resulting image when preview is blurry.

**Expected Behaviour**
- Auto focus set to off
- Touch on camera preview, the preview focus on the image correctly
- Press a button to call takePicture
- The camera preview focus based on the autoFocusPointOfInterest coordinate, no more focusing happened
- The resulting image when preview is clear and focused.

**Steps to Reproduce**
Follow the behaviour above

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

bobmulder, 2019-06-16:
I do have this issue for a long time on Xiaomi phone. Any idea how this happens?

andresmtz98, 2019-08-06:
Hey bro, how did you do to focus on touch? I've looking that functionality but nothing :/

cristianoccazinsp, 2019-08-08:
Having the same issue. I'm setting the ```autoFocusPointOfInterest``` prop to focus on a specific point of the image. The preview looks fine; however, focus is quickly lost after calling takePictureAsync.

Overall, the focus is lost very quickly as well.

cristianoccazinsp, 2019-10-07:
I don't think this is an issue anymore with Camera1 implementation.

castalonirenz, 2020-08-07:
How to implement touch to focus

closed 2021-01-06 by stale[bot]