react-native-camera/react-native-camera #2310: Best way to record 20 second interval videos

charlesinwald, 2019-06-10:
# Question

**To Do First**
- [X] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [X] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [X] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

I'm trying to record a new video every 20 seconds, and then decide whether to save it based on external criteria.  So far I have maxDuration option set to 20 and have the following code:
```
    takeVideo = async (camera) => {
        console.log('Top of takeVideo');
        try {
            const promise = camera.recordAsync(this.state.recordOptions);
            if (promise) {
                console.log('isRecording');
                const data = await promise;
                console.log('After promise');
                console.warn('takeVideo', data);
            }
        } catch (e) {
            console.error(e);
        }
    };

    captureVideoUnit(camera) {
        setInterval(console.log('Start of interval'),20000);
        setInterval(this.takeVideo(camera), 20000);
        if(camera) {
            console.log('about to stop');
            camera.stopRecording();
        }

    }
```
I call `this.captureVideoUnit(camera);` just before the return of the render function.  I am using the Facc (Function as Child Components) implementation.

I have been toying with this code, and I'll either get an "undefined is not an object" error or it will constantly start/stop and crash.  What is the best way to implement an interval?
**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

sibelius, 2019-06-27:
Try to keep the same path https://github.com/react-native-community/react-native-camera/blob/master/src/RNCamera.js#L226

if you want to record the video in  the same file

or use different paths

try to use hooks to make your logic easy to reason about it

charlesinwald, 2019-06-27:
I'm a bit confused as to what you are referring to, I'm wondering how to record video on a timed interval.

sibelius, 2019-06-27:
start and stop at each interval

charlesinwald, 2019-06-27:
That I understand, but with FaCC, how do I pass the camera object to the timer?

sibelius, 2019-06-27:
save on a ref

const cameraRef = useRef()

<Camera ref={(ref) => cameraRef.ref = ref>

closed 2019-09-02 by stale[bot]

leonidlewis, 2021-11-18:
I have encountered a similar issue. I want to record the video in 30 seconds but when I start recording, the app crashes.