react-native-camera/react-native-camera #2312: migrate from v0.x

Symous, 2019-06-12:
# Question

**To Do First**
- [x] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [x] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [x] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**

Hi, I integrated this lib(v.0.6.0) to my RN app years ago, and I have found many problems that I can not handle due to the outdate of the version, that drive me crazy, I just want to know is there any way that I can upgrade to 2.0 with the minimum cost？Thanks.
**Tags**

MateusAndrade, 2019-06-12:
Upgrading with a "minimal cost" is a thought question based on the version you are using. Is your project a simple app? 

Please give a look at: https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md.

Btw, im closing this issue, but you can comments any doubt/problem you are having. =D

closed 2019-06-12 by MateusAndrade

Symous, 2019-06-15:
@MateusAndrade Thanks my friend. That's a really hard work for me to migrate from the very old version.
For someone who are looking for the solution:
1. unlink and uninstall the old react-native-camera
2. install and link the latest react-native-camera
3. check the [migration doc](https://github.com/react-native-community/react-native-camera/blob/master/docs/migration.md) and do some manually configuration(**especially Android gradle**). [this](https://github.com/react-native-community/react-native-camera/blob/master/docs/migrationV2.md) may also help.
5. Modify the RCTCamera API to RNCamera API.
6. Handle some collision with third party libs(for me I have to make modifications to react-native-code-push due to the upgrade of Android Gradle ).