react-native-camera/react-native-camera #2315: Reopen: A way to read data byte from current preview, like a previewCallback?

DennisJu007, 2019-06-13:
# Feature-Request

**Describe the Feature**
Is there a way to read the data from the current preview? My goal is to retrieve the camera preview as a Pixel Array.

A preview callback would be a good addition, passing the data byte, just like the onPreviewFrame callback when you set it up.

**Possible Implementations**

I need the byte array of onPreviewFrame(byte[], Camera) to change to rgb values.
[image]

**Related Issues**

#451

n1ru4l, 2019-06-13:
This might be a huge performance hit, I do not think sending a whole picture via the javascript bridge is a good idea.

DennisJu007, 2019-06-13:
I cant agree with you more.Performance is very important for us. Maybe be we can find another way to resolve it.We can support a function which needs a  screen's point x, y params and returns this point's rgb values. How does it sound like?

closed 2019-08-19 by stale[bot]