react-native-camera/react-native-camera #2316: Cannot build on iOS due to missing RNFaceEncoder.m

laurent22, 2019-06-14:
I'm struggling to get RNCamera to build on iOS. I couldn't find any installation instructions, so I've installed it like most RN modules:

    npm i -s react-native-camera
    react-native link react-native-camera

However I'm getting the following error:

> error: Build input file cannot be found: 'node_modules/react-native-camera/ios/FaceDetector/RNFaceEncoder.m'

And indeed when I search for this file it's not present in that location (and also not in this repository).

I'm probably missing something obvious but not sure what. What would be the way to get this working?

sibelius, 2019-06-14:
you need to use cocoapods

laurent22, 2019-06-14:
Thanks but I can't use Cocoapods unfortunately as I'd have to change the way all dependencies are managed, which would probably bring a lot of new problems.

Eventually I found the [installation instructions](https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md) and manual installed it. Anything else would not work for me.

closed 2019-06-14 by laurent22

viananicolas, 2019-06-26:
@laurent22, what exact steps did you follow? Tried using CocoaPods, but it just let me down to a rabbit hole of many new errors, and I'm at a total loss.

sibelius, 2019-06-26:
try to use autolink with cocoapods, it really helps

laurent22, 2019-06-26:
@viananicolas, manual iOS install is what worked for me, just follow the steps here https://github.com/react-native-community/react-native-camera/blob/master/docs/installation.md#ios

viananicolas, 2019-06-27:
Thanks, guys. Turns out that I needed to remove and add again `RNCamera.xcodeproj` in **Libraries** and `libRNCamera.a` in **Link Binary With Libraries**.

howieyoung, 2019-08-27:
Yeah, relink the libraries manually should fix the issue.