react-native-camera/react-native-camera #2319: Support UPC-A barcodes

castillobgr, 2019-06-18:
# Feature-Request

**Describe the Feature**

It would be awesome for `react-native-camera` to support UPC-A barcodes ([example](https://barcode.tec-it.com/en/UPCA?data=72527273070)). Right now they're being detected as EAN-13, but with a `0` prepended to the code.

**Possible Implementations**

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

castillobgr, 2019-06-18:
Upon digging a little further, I see that UPC-A is [listed as one of the valid types in the Android native module](https://github.com/react-native-community/react-native-camera/blob/22533ed8e8c47f17ed7ca28f5879064fc2572a7f/android/src/main/java/org/reactnative/camera/CameraModule.java#L61), but not [in Typescript](https://github.com/react-native-community/react-native-camera/blob/master/types/index.d.ts#L45). Why is this?

castillobgr, 2019-06-19:
I switched to using `onGoogleVisionBarcodesDetected` instead, and it supports UPC-A barcodes. Closing this.

CorentinHeroux, 2020-05-29:
@0x5d And for iOS ?

0x5d, 2020-05-29:
@CorentinHeroux this was a year ago, so I can't check, but we didn't need to support iOS at the time.

closed 2020-08-05 by stale[bot]

JuanCamilo0809, 2020-12-14:
I need help with this issue.