react-native-camera/react-native-camera #2320: java.lang.RuntimeException: Manifest merger failed with multiple errors, see logs

xgAnd, 2019-06-19:
# Bug Report
when use react-native-camera get the error
**To Do First**

- [ ] Did you try latest release? 
use the version 2.9.0 still error
- [ ] Did you try master?
No
- [ ] Did you look for existing matching issues? 
yes
**Platforms**

**Versions**

- Android:
        buildToolsVersion = "28.0.2"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 27
        supportLibVersion = "28.0.0"
- iOS:
- react-native-camera:
2.9.0 and 2.2.0 both has the error
- react-native:
0.58.6
- react:
16.6.3
  

**Description/Current Behaviour**
use this module  get the error Manifest merger failed with multiple errors. i use ./gradlew app:dependencies find this module use the androidx info .i see the PR said update androidstudio and gradle to the latest but the error is still exist . when i remove this module the error resolved so i think error is from this.
the module react-native-device-info contain androidx also. i can use 
configurations.all {
    resolutionStrategy.force 'com.google.android.gms:play-services-gcm:16.1.0', 'com.google.android.gms:play-services-basement:15.0.1'

}
to resolved. i try to download com.google.firebase:firebase-ml-version: other version but failed. how to fixed this error? look forward to your reply.thanks.
[image]

**Expected Behaviour** has no error

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

UADACID, 2019-06-19:
i am still waiting for solving this issue

hntddt123, 2019-06-19:
You can solve it by 
1. Going into node_module/react-native-camera for each not exist error java files and change old android import to androidx import.
2. Add two lines to 
```android.useAndroidX=true```
```android.enableJetifier=true```
in gradle.properties.

I am using react-native-camera: 2.10.2

xgAnd, 2019-06-20:
> You can solve it by
> 
> 1. Going into node_module/react-native-camera for each not exist error in java files and change old android import to androidx import.
>    https://developer.android.com/jetpack/androidx/migrate
> 2. Add two lines to
>    `android.useAndroidX=true`
>    `android.enableJetifier=true`
>    in gradle.properties.
> 3. Under build.gradle you may need to add these lines
> 
> ```
> dependencies {
>     implementation 'androidx.appcompat:appcompat:1.0.2'
>     implementation 'androidx.annotation:annotation:1.1.0'
>     implementation 'androidx.legacy:legacy-support-v4:1.0.0'
> }
> ```
> 
> 1. Under AndroidManifest.xml add this line inside <application tag to make react-native work
> 
> ```
> android:usesCleartextTraffic="true"
> ```
> 
> I am using react-native-camera: 2.10.2

hello,thanks for reply.i see you suggest mean let me change my project to merge androidx. but we peoject ues a lot of third model and some module the author is not maintain now,so if merge to androidx is a not easy work and time is not enough.so i hope have some method
```
configurations.all {
resolutionStrategy.force 'com.google.android.gms:play-services-gcm:16.1.0', 'com.google.android.gms:play-services-basement:15.0.1'
}
```
like this can change the firebase-ml-version do not merge to androidx for the time.

hntddt123, 2019-06-20:
I changed files to androidx in node_module/react-native-camera for my project and it works without changing the implementation part. Only import changes
AspectRatio.java,
CameraView.java,
CameraViewManager.java,
Camera1.java,
Camera2.java,
FileFaceDetectionAsyncTask.java,
MutableImage.java,
Size.java,
SizeMap.java,
RNCameraView.java,
RNCameraViewHelper.java,
RCTCameraViewManager.java,
ResolveTakenPictureAsyncTask.java

mikehardy, 2019-06-20:
f you must have androidx support, use https://github.com/mikehardy/jetifier in a package.json postinstall hook. react-native-camera is one of the dependencies I test the package with

https://github.com/mikehardy/rn-androidx-demo/blob/master/make-demo.sh#L12

mikehardy, 2019-06-20:
Worth mentioning maybe that as I test jetifier I found it was not possible to pin the mlkit gradle dependencies to specific versions because gradle can't handle variables with '-' in them, so you can't specify ml-vision-version etc in android/build.gradle - this will get people later

closed 2019-08-26 by stale[bot]

waed12, 2021-11-05:
[image]

LiFeng1783492844, 2023-06-01:
> hello,thanks for reply.i see you suggest mean let me change my project to merge androidx. but we peoject ues a lot of third model and some module the author is not maintain now,so if merge to [...]

**I am using 2.6.0. my project cannot merge to androidx.
I resolve it just by remove androidx.exifinterface**
> configurations.all {
>     exclude group: 'androidx.exifinterface', module: 'exifinterface'
> }
 because react-native-camera 2.6.0 has implementation "com.android.support:exifinterface:${safeExtGet('supportLibVersion', '28.0.0')}"

[image]