react-native-camera/react-native-camera #2321: Version 2.10.1 reverted AndroidX migration.

tomaszrybakiewicz, 2019-06-19:
# Bug Report

With version 2.10.0  react-native-camera migrated from using android support libraries to AndroidX. However it appears that with version 2.10.1 this has been reverted. Probably one of the pre 2.10.0 Pull Requests that got merged without proper code review.

Here's one of the files affected.
_from v2.10.0_
https://github.com/react-native-community/react-native-camera/blob/v2.10.0/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java

_from v2.10.1_
https://github.com/react-native-community/react-native-camera/blob/v2.10.1/android/src/main/java/org/reactnative/camera/RNCameraViewHelper.java

mikehardy, 2019-06-20:
It was reverted because it's a breaking change, it should have been released in a 3.0

If you must have androidx support, use https://github.com/mikehardy/jetifier in a package.json postinstall hook. react-native-camera is one of the dependencies I test the package with 

https://github.com/mikehardy/rn-androidx-demo/blob/master/make-demo.sh#L12

tomaszrybakiewicz, 2019-06-21:
Gotcha. Got excited when saw migration in release notes for 2.10.0. Didn't notice that 2.10.1 reverted it. My bad. Thanks for the update.

closed 2019-06-21 by tomaszrybakiewicz