react-native-camera/react-native-camera #2323: Empty camera permissions rationale popup

castillobgr, 2019-06-19:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android: 9
- iOS:
- react-native-camera: 2.10.2
- react-native: 0.59.4
- react: 16.8.3
- react-navigation: 3.6.1

**Description/Current Behaviour**
When rendering an `RNCamera` component without `androidCameraPermissionOptions`, a blank rationale popup will be rendered before the permissions dialog. It only goes away by tapping anywhere else in the screen, or by pressing the OS back button, after which the permissions dialog is shown. (Sometimes this only happens after the permissions are initially denied, and the user retries).

**Expected Behaviour**
If no `androidCameraPermissionOptions` are passed, `RNCamera` should go straight to showing the permissions dialog.

**Steps to Reproduce**

See the **Description** section.

**Additionals**
Screenshot:

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

castillobgr, 2019-06-19:
I looked around and found this: https://github.com/react-native-community/react-native-camera/blob/8ad809ee889ebcdcf0c0b4b891f423f2354a1dd8/src/RNCamera.js#L426

Perhaps `androidCameraPermissionOptions` should be `undefined` in `defaultProps` since it's optional?

MateusAndrade, 2019-06-20:
I gave a quick look at RNCamera.js and it looks to be related to this function:

https://github.com/react-native-community/react-native-camera/blob/8ad809ee889ebcdcf0c0b4b891f423f2354a1dd8/src/RNCamera.js#L39&L42.

I'll try to give more attention to this over the weekend. Btw, thanks for describing your issue following our template. :smile:

castillobgr, 2019-06-20:
Thank you for replying!

Yup, that's why I mentioned the object's optionality and setting it in `defaultProps` as `undefined`. If `PermissionAndroid.request`'s 2nd argument is omitted, it doesn't show the rationale popup.

MateusAndrade, 2019-06-22:
Im debugging this, and found that is something related to this: https://github.com/react-native-community/react-native-camera/blob/8ad809ee889ebcdcf0c0b4b891f423f2354a1dd8/src/RNCamera.js#L27&L73

If you apply a any value to ```androidCameraPermissionOptions```, something similar to this: 

```javascript

    if (!cameraPermissions.message) {
      cameraPermissions = {
        ...cameraPermissions,
        message: 'Some default message',
      };
    }

    if (!cameraPermissions.title) {
      cameraPermissions = {
        ...cameraPermissions,
        title: 'Some default title',
      };
    }
```

You will get this: 

[image]

I will try getting more info about this, but fell free to send a PR if you find whats happening. :smile:

MateusAndrade, 2019-06-22:
To use this on your app without this empty PopUp just pass this prop to RNCamera.

```javascript
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
```

By some reason, it will render the PopUp with the options you passed, after the that the Android permission will be asked.

[image]

Im running out of time at this weekend, but it looks related to: https://facebook.github.io/react-native/docs/permissionsandroid behaviour

bwor, 2019-07-11:
May I ask if this problem has been solved

MateusAndrade, 2019-07-15:
@bwor i was looking at this, but i dont have time on the last weeks to give this the proper attention. If you still have this issue, simple pass this: 

```javascript
   androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
   }}
```

Thsi will prevent the empty reation from appearing to your screen, showing a dialog with the labels that you pass.

Symous, 2019-09-02:
Not solved even I add androidCameraPermissionOptions prop to RNCamera.

gauravahujame, 2019-10-03:
You can pass null in androidCameraPermissionOptions to avoid the empty dialog box. Same goes for androidRecordAudioPermissionOptions.
`
androidCameraPermissionOptions={null}
androidRecordAudioPermissionOptions={null}
`

closed 2019-12-09 by stale[bot]

MateusAndrade, 2020-03-13:
fixed on #2740.