react-native-camera/react-native-camera #2325: Could not resolve all files for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'

kgrajan12, 2019-06-20:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
Anyone help me in this problem
I have installed the react-native-camera and linked generally then it shows like this

> Task :react-native-camera:generateMlkitDebugRFile FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Could not resolve all files for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'.
> Failed to transform file 'bc405fd5ca5ef6a20e8d11944b4be704' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'cf1023f7f85d7273f58ff3d6dcf2a8ea' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file '74c5247665cf1b59b38652e31d28311e' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'e95c068a565b007332239cea3b17db09' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'bea33a9c27253b0c696272c07c0aa516' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file '10bf2bcac9a1a076f1c0e4f533689b24' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'c78b42fd8271a578580316d0faaa588f' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'c78b42fd8271a578580316d0faaa588f' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file 'c78b42fd8271a578580316d0faaa588f' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.
> Failed to transform file '297515b8a8a97b0e0e35c5e87147031b' to match attributes {artifactType=android-symbol-with-package-name} using transform LibrarySymbolTableTransform
   > java.io.IOException: org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 1; Content is not allowed in prolog.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 3m 2s
315 actionable tasks: 1 executed, 314 up-to-date

**Tags**
### Environment
React Native Environment Info:
    System:
      OS: Windows 10
      CPU: (2) x64 Intel(R) Core(TM)2 Duo CPU     E7500  @ 2.93GHz
      Memory: 1.37 GB / 3.97 GB
    Binaries:
      Yarn: 1.16.0 - C:\Users\Gajarajan\AppData\Roaming\npm\yarn.CMD
      npm: 5.6.0 - C:\Program Files\nodejs\npm.CMD
    IDEs:
      Android Studio: Version  3.4.0.0 AI-183.6156.11.34.5522156

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-06-22:
@kgrajan12, you can compare your code with the MLKit to see if you can find any missing part? 

https://github.com/react-native-community/react-native-camera/tree/master/examples/mlkit

kgrajan12, 2019-06-24:
But I don't use mlkit in this package

kgrajan12, 2019-06-24:
And I solved this
Delete all the files in cache
C:/user/USERNAME/.gradle/cache

closed 2019-06-24 by kgrajan12