react-native-camera/react-native-camera #2333: Error when compiling android

andcosta, 2019-06-23:
**package.js**

```
...
"react": "16.8.3",
"react-native": "0.59.5",
"react-native-camera": "git+https://git@github.com/react-native-community/react-native-camera.git"
...
```

**settings.gradle**

```
include ':app'

include ':react-native-camera'
project(':react-native-camera').projectDir = new File(rootProject.projectDir, '../node_modules/react-native-camera/android')
```

**build.gradle**

```
...
android {
    compileSdkVersion rootProject.ext.compileSdkVersion

    compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }

    defaultConfig {
        applicationId "xxxxxx"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        versionCode 1
        versionName "1.0"
    }
    ...
}

dependencies {
    implementation project(':react-native-camera')
    implementation fileTree(dir: "libs", include: ["*.jar"])
    implementation "com.android.support:appcompat-v7:${rootProject.ext.supportLibVersion}"
    implementation "com.facebook.react:react-native:+"  // From node_modules
}
```

**Error**

../../../../../../../../node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java (3 errors)

error: cannot find symbol class NonNull
error: cannot find symbol class NonNull
error: cannot find symbol class NonNull

gitpayal, 2019-06-24:
I am having same issue.I did migration to AndroidX from Android Studio( v3.4.1) but the issue persists

hntddt123, 2019-06-24:
see #2320

closed 2019-06-28 by andcosta