react-native-camera/react-native-camera #2334: onTextRecognized crash on android

mengheangrat, 2019-06-24:
# Bug Report

**To Do First**

**Platforms**

-Android

**Versions**

- Android:  Pie
- react-native-camera: 2.11.0
- react-native :0.59.3
- react: 16.8.3

**Description/Current Behaviour**

- Use text recognize 
- Start open camera 
- got crash 

`06-24 17:37:52.727 6458-6490/com.testsaga E/AndroidRuntime: FATAL EXCEPTION: AsyncTask #3
    Process: com.testsaga, PID: 6458
    java.lang.RuntimeException: An error occurred while executing doInBackground()`

closed 2019-06-24 by mengheangrat

pedrosimao, 2019-06-28:
I have the same issue. Did you find a solution @mengheangrat ? Can you please share it here? Thanks!

pedrosimao, 2019-06-28:
Ok, I have found my culprit.
on `build.gradle` file I have changed this:
```implementation(project(":react-native-device-info"),  {
        exclude group: "com.google.android.gms"
    })
```

to this:
` implementation project(":react-native-device-info")`

the exclude is needed for other libs that generate a conflict