react-native-camera/react-native-camera #2336: Android Build failed for task ':react-native-camera:compileGeneralReleaseJavaWithJavac'.

Aromxyz, 2019-06-25:
# What went wrong:
    Execution failed for task ':react-native-camera:compileGeneralReleaseJavaWithJavac'.

# Steps to reproduce
1. ./gradlew clean && ./gradlew assembleRelease

# Environment
1. node.js version: 10.11.0
2. react native version: 0.59

# react-native-camera
Version: npm version or "master"

Aromxyz, 2019-06-25:
my android/build.gradle : `buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 28
        supportLibVersion = "28.0.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.3.1'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        //maven {url 'https://dl.bintray.com/jetbrains/anko'}
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-vision:11.8.0'
                force 'com.google.android.gms:play-services-gcm:11.8.0'
            }
        }

    }
}`

Aromxyz, 2019-06-25:
My android/build.gradle 
``
buildscript {
    ext {
        buildToolsVersion = "28.0.3"
        minSdkVersion = 16
        compileSdkVersion = 28
        targetSdkVersion = 28
        supportLibVersion = "28.0.0"
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath 'com.android.tools.build:gradle:3.3.1'

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}

allprojects {
    repositories {
        mavenLocal()
        google()
        jcenter()
        //maven {url 'https://dl.bintray.com/jetbrains/anko'}
        maven {
            // All of React Native (JS, Obj-C sources, Android binaries) is installed from npm
            url "$rootDir/../node_modules/react-native/android"
        }
        configurations.all {
            resolutionStrategy {
                force 'com.google.android.gms:play-services-vision:11.8.0'
                force 'com.google.android.gms:play-services-gcm:11.8.0'
            }
        }

    }
}``

Tino-F, 2019-06-26:
Also having the same issue.

Tino-F, 2019-06-26:
I see that you're using [react-native-device-info](https://github.com/react-native-community/react-native-device-info), so it's probably related to https://github.com/react-native-community/react-native-camera/issues/2240

closed 2019-09-01 by stale[bot]

tahmidrahman-dsi, 2019-12-22:
@Aromxyz did you find any solution?