react-native-camera/react-native-camera #2337: NO Error And Black Screen on Android

GospelHW, 2019-06-25:
# Bug Report
Any body help me?
**To Do First**

- [yes] Did you try latest release?
- [no] Did you try master?
- [yes] Did you look for existing matching issues?

**Platforms**

Android 

**Versions**

- Android:28
- iOS:12.3.1
- react-native-camera:2.11.0
- react-native:0.59.8
- react:16.8
  

**Description/Current Behaviour**

I run in Android ,I just see black screen 

**Expected Behaviour**

**Steps to Reproduce**

1、npm install react-native-camera --save
2、react-native link react-native-camera
3、npm install react-native-qrcode-scanner --save
4、react-native link react-native-qrcode-scanner
5、react-native link react-native-permissions
6、react-naitve run-android

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2019-06-25:
You applied the permission tô access the câmera on the manifest?

GospelHW, 2019-06-25:
@MateusAndrade Yes I'm sure.

MateusAndrade, 2019-06-25:
Maybe it can be related to #2252 .

Btw, can provide the repo with your code?

iagorm, 2019-06-26:
This is happening with me in Android, by using only those code:

```
<RNCamera
        ref={(camera) => { this.camera = camera; }}
        androidCameraPermissionOptions={{
          title: 'Permission to use camera',
          message: 'We need your permission to use your camera',
          buttonPositive: 'Ok',
          buttonNegative: 'Cancel',
        }}
        captureAudio={false}
        style={{ flex: 1 }}
        type={RNCamera.Constants.Type.front}
      >
      <View style={{flex: 1}/>
      </RNCamera>
```

To reproduce the error: open a screen with this component, leave and enter again and the screen will not show

ajmalsalim, 2019-06-27:
I have the same issue on Android. Tried to move from PureComponent to Component but it is not fixing for me.

wellington-cunha, 2019-08-16:
I solved my problem using  "react-native-camera": "1.2.0",

ntk0104, 2019-09-13:
> @ntk0104 In my case was just remove the aspect ratio prop from component, I guess it is causing conflict.

can you explain more detail ? Thanks

ntk0104, 2019-09-13:
I have fixed it by the solution that i found from another issue post.
`renderCamera = () => {
    const isActive = this.props.navigation.isFocused()
    if(isActive == true){
      return(
        <RNCamera ref={ref => { this.camera = ref }}
        onBarCodeRead={e => this._handleScanCode(e)}
        style={styles.preview}>
        ...
      </RNCamera>
      )
    }else {
      return null
    }
  }
  };`

In my case, i use componentShouldUpdate and only allow re-render when this variable this.props.navigation.isFocused == true, so this stupid things take my whole day to solved this problem. You should make the UI re-render when the variable this.props.navigation.isFocused change like as: 
true => mount and render the camera
false => unmount and render null
This solution works perfectly with me. Gud luck

wellington-cunha, 2019-09-13:
@GospelHW GospelHW 现在行了？

MaximilianoRG, 2019-09-17:
IMPORTANT: 
In ios folder , execute the command "pod install" after linking manually the library

kevtran410, 2019-10-17:
I got the same issue today without changing anything of code (before work fine)
Anyone solved this problem ?
Version i used:
- react-natve 0.59.4
- react-native-navigation: 2.21.0
- react-native-camera: 2.11.2

Janjiran, 2019-10-29:
Same issue today.

- react 0.61.2
- react-native-camera 3.0.1
- react 16.0.9
- Android API 27 (8.1)
- Mac os 10.15
- Node V10.17.0

In android studio console, there is error:
`W/CameraBase: An error occurred while connecting to camera 0: Status(-8): '8: connectHelper:1305: Cannot open camera 0 for "<AppName>" (PID -1): Too many other clients connecting'`

When camera is open, whole screen is just black.
running app with multidex enabled, using `react-navigation 4.0.10`, `react-navigation-stack 2.0.0-alpha.7` and `react-native-gesture-handle 1.4.1` but i tried to use the example app in top `index.js` file so navigation is not even initialized.
I'v also tried solution from @ntk0104 but it's also not working.

I'm trying to solve this whole day and still same result.

doubtful-user, 2019-10-29:
Please :(

StannyS, 2019-10-29:
+

StannyS, 2019-10-29:
+

codervss, 2019-10-30:
Have tried almost everything, but the error still persists.
Someone Please help us out

daxaxelrod, 2019-10-31:
Having the same issues, See below from a snippet from logcat
2019-10-30 20:03:08.396 22107-22303/? E/QCamera: <HAL><ERROR> openCamera: 2442: camera_open failed. rc = -87, mCameraHandle = 0x0
2019-10-30 20:03:08.396 717-812/? E/ANDR-PERF-MPCTL: Invalid profile no. 0, total profiles 0 only
2019-10-30 20:03:08.399 22107-22303/? I/QCameraHalWatchdog: Stopped Watchdog Thread (3ms)[openCamera]
2019-10-30 20:03:08.399 22107-22303/? I/QCamera: <HAL><INFO> openCamera: 2367: [KPI Perf]: X PROFILE_OPEN_CAMERA camera id 0, rc: -87
2019-10-30 20:03:08.401 22107-22303/? I/QCamera: <HAL><INFO> closeCamera: 2712: E
2019-10-30 20:03:08.403 22107-22303/? E/MOTCamDev@1.0-impl: Could not open camera 0: -87
2019-10-30 20:03:08.403 1015-1015/? E/CameraHardwareInterface: initialize: failed to get the bgproc instance
2019-10-30 20:03:08.403 1015-1015/? E/CameraHardwareInterface: initialize: openSession failed! Too many users (-87)
2019-10-30 20:03:08.403 1015-1015/? E/CameraClient: initialize: Camera 0: unable to initialize device: Too many users (-87)
2019-10-30 20:03:08.403 1015-1015/? I/CameraHardwareInterface: Destroying camera 0
2019-10-30 20:03:08.404 1015-1015/? E/CameraService: connectHelper: Could not initialize client from HAL.

Janjiran, 2019-11-05:
For anybody facing the same issue. Since nobody cares about this issue, I was forced to use react-native-camera-kit but I have to say it's working pretty damn good, installation is outdated for androidX support, but you can find guide in issues.

michael-harley, 2019-11-07:
Still the same issue:

### Dependency
- react: 16.9.0
- react-native: 0.61.4
- react-native-camera: 3.9.0
- react-navigation: 4.0.5
- react-navigation-stack: 1.8.0

### Happens when:
1. First opening <RNCamera /> : **Success**
2. Press back button
3. Open <RNCamera /> again: **Failed** (Component keeps showing loading with white overlay)

### Have tried:
**First Solution**
Using react-navigation `isFocused` and `withNavigationFocus` HOC.
- Apply `isFocused` to render
```
isFocused && <RNCamera />
```
or
```
isFocused ? <RNCamera /> : null
```
or
```
shouldComponentUpdate(nextProps, nextState){
    const { isFocused } = this.props;
    return isFocused;
}
```
All failed.

**Second Solution**
Try to change back button to Navigation Reset instead.
```
handleBackPress = () => {
    // this.homeButtonReset()
    const {featureIds, user} = this.props.navigation.state.params;
    const {dispatch} = this.props.navigation;
    const replace = StackActions.reset({
      index: 0,
      actions: [
        NavigationActions.navigate({
          routeName: 'InputHome',
          params: {
            featureIds,
            user,
          },
          key: null,
        }),
      ],
    });
    dispatch(replace);
    return true;
  };

UNSAFE_componentWillMount() {
    BackHandler.addEventListener('hardwareBackPress', this.handleBackPress);
  }
```
Still failed.

-----
Tried to solve this for 2 days, still no good news.

michael-harley, 2019-11-08:
I have found the solution guys.

For those who still have problem regarding this, please change your `react-native-camera` dependency to version `3.8.0`.

You can check [here](https://www.npmjs.com/package/react-native-camera), and see a pattern that the error occurs for these few days. (4-5 days)

And that is exactly the day the error occurred.
I'm using the exact same version as before, but only changing the `react-native-camera` version to 3.8.0.

It works for me, might be able to help you.
Good luck.

(I've seen that your react-native-camera version varies, but might worth the time to try this solution)

michael-harley, 2019-11-08:
I have found the solution guys.

For those who still have problem regarding this, please change your `react-native-camera` dependency to version `3.8.0`.

You can check [here](https://www.npmjs.com/package/react-native-camera), and see a pattern that the error occurs for these few days. (4-5 days)

And that is exactly the day the version is changed.
I'm using the exact same version as before, but only changing the `react-native-camera` version to 3.8.0.

It works for me, might be able to help you.
Good luck.

(I've seen that your react-native-camera version varies, but might worth the time to try this solution)

michael-harley, 2019-11-08:
I have found the solution guys.

For those who still have problem regarding this, please change your `react-native-camera` dependency to version `3.8.0`.

You can check [here](https://www.npmjs.com/package/react-native-camera), and see a pattern that the error occurs for these few days. (4-5 days)

And that is exactly the day the version is changed. (3.9.0)
I'm using the exact same version as before, but only changing the `react-native-camera` version to 3.8.0.

It works for me, might be able to help you.
Good luck.

(I've seen that your `react-native-camera` version varies, but might worth the time to try this solution)

Pingou, 2019-11-14:
Do you guys use react-native-screens?

Janjiran, 2019-11-14:
> Do you guys use react-native-screens?

Yes. `2.0.0-alpha.4`

kevtran410, 2019-11-26:
After the last few days, we found bugs in our projects.
Set `android:hardwareAccelerated="false"` in `AndroidManifest.xml` makes the camera no longer display.
Check your AndroidManifest and set it back to true may help the camera work again.

alexsorokoletov, 2020-01-09:
> Do you guys use react-native-screens?
> I would suggest to try to disable it if you can and see if that helps.

Why so?

o-pooja-o, 2020-01-31:
If you are passing AppContainer in your App.js file, dont wrap it in View Or blank tag.
Screen won't be blank anymore :D

vauvenal5, 2020-03-02:
@o-pooja-o do you know the reason for this behavior?

dangkhoa2708, 2020-03-07:
Still happening ? Any solution yet guys ?

o-pooja-o, 2020-03-16:
@dangkhoa2708 
probably its because AppContainer is not an actual component you wanted as a view in your app. 
AppContainer contains navigation functionalities which are not supposed to show in frontend of your app. It does work in background. View tag descibes how inside component should be displayed.

o-pooja-o, 2020-03-16:
> Still happening ? Any solution yet guys ?

@vauvenal5 try this command on terminal while you run app
adb logcat.
It will show the errors happening behind the white screen probably or whats happening wrong with your app.

o-pooja-o, 2020-03-16:
> @o-pooja-o do you know the reason for this behavior?

probably its because AppContainer is not an actual component you wanted as a view in your app. 
AppContainer contains navigation functionalities which are not supposed to show in frontend of your app. It does work in background. View tag descibes how inside component should be displayed.

o-pooja-o, 2020-03-16:
> Still happening ? Any solution yet guys ?

@dangkhoa2708 
 try this command on terminal while you run app
adb logcat.
It will show the errors happening behind the white screen probably or whats happening wrong with your app.

o-pooja-o, 2020-03-16:
> @o-pooja-o do you know the reason for this behavior?

@vauvenal5 
probably its because AppContainer is not an actual component you wanted as a view in your app. 
AppContainer contains navigation functionalities which are not supposed to show in frontend of your app. It does work in background. View tag descibes how inside component should be displayed.

anwarhamr, 2020-03-30:
I also struggled w/ this and tried all the things above...  The app worked fine with android 9 but on android 6 (Marshmallow) I was getting the black screen.
@o-pooja-o 's recommendation to use logcat to figure it out helped me.  To do that I did the following:

1.  Restart the device
2.  Stopped the app on the device
3.  Ensure the device is connected `adb devices`
4.  Delete the log `adb logcat -c`
5.  Save log to a file `adb logcat >> myDeviceOut.txt`
6.  Start the application and go to the screen where the camera should work.
7.  Open up the output file in a text editor and search for the string 'camera'; look at all the entries and try to determine what has happened.

I'm running rn 60.5, rncamera 3.15.1, have the permissions request correct and using pureComponents, View at top level and just about everything people were saying to fix/remove...

My log had a message that the Camera Service had crashed.... with no way to recover I restarted the device.  This went on an on until I had the revelation to just open up the OS's Camera application and try to take a couple pictures.  This didn't work either because I had my app run prior, which may have crashed the camera service again... 
So I did one more device restart, opened default camera app installed on the device and took a few pics, worked great.  I then opened up my app went to my camera screen.  And wouldn't you know the damn camera never loaded, not even a black screen. Because the default camera application was still active in the background I'm guessing my application could not get access from the OS.  I closed both applications and reopened mine and shazzam camera is working!!!  
I sure hope this helps someone.  Happy Hacking.

khyper, 2020-05-14:
> After the last few days, we found bugs in our projects.
> Set `android:hardwareAccelerated="�false"` in `AndroidManifest.xml` makes the camera no longer display.
> Check your AndroidManifest and set it back to true may help the camera work again.

thanks @kevtran410, the camera is working now

pke, 2020-05-15:
For me the example code just does not work. The style of the preview must have `alignSelf: "stretch` to show the image. The camera was working as the indicator on the phone showed, but the preview did not enlarge in the parent view (as per example code).

```js
return <View style={{ flex: 1, backgroundColor: "black" }}>
  <RNCamera style={{ flex: 1, alignSelf: "stretch" }}
    captureAudio={false} 
    type={RNCamera.Constants.Type.back}
    flashMode={RNCamera.Constants.FlashMode.on}
    androidCameraPermissionOptions={{
      title: "Permission to use camera",
      message: "We need your permission to use your camera",
      buttonPositive: "Ok",
      buttonNegative: "Cancel",
     }}
  />
</View>
```

Gastonovich, 2020-06-26:
Hi guys, trying to fix this issue all day 
My problem was that camera component located in nested navigation stack, putting it at top level of navigation resolve helped

closed 2020-09-02 by stale[bot]

raazatul7, 2020-11-02:
Still facing the same issue in the Samsung device.

tayloraleach, 2021-02-04:
This is going to sound really stupid but my brand new react native 0.63 project just playing around with this lib and my camera was working great until all of a sudden black screen and the app locked up and would not load on my physical android device. Tried reinstalling, cleaning, restarting servers and nothing worked. 

Then I somehow noticed on my physical device that my flashlight toggle would not work - even with the app UNINSTALLED... 

So I rebooted my phone and it works again. Not sure how it got into this state but might be worth ruling out a glitch in the matrix? :)

ovinokurov, 2021-02-07:
This is a very easy fix, I don't understand why nobody came up with the solution before. 

1) Go to Android Studio
2) Open your [Android Virtual Device Manager] in [Your Virtual Devices] delete all devices and add a new one.

Now Run :) good luck!

dhavalpanchani, 2021-02-17:
> After the last few days, we found bugs in our projects.
> Set `android:hardwareAccelerated="�false"` in `AndroidManifest.xml` makes the camera no longer display.
> Check your AndroidManifest and set it back to true may help the camera work again.

This worked for me.

cmeiller, 2021-03-24:
Same issue here on Huawei device. Black screen on first render...

> After the last few days, we found bugs in our projects.
> Set `android:hardwareAccelerated="�false"` in `AndroidManifest.xml` makes the camera no longer display.
> Check your AndroidManifest and set it back to true may help the camera work again.

Nothing changes with that.
Any solution? 🙏

s10mcow, 2021-07-29:
> This is going to sound really stupid but my brand new react native 0.63 project just playing around with this lib and my camera was working great until all of a sudden black screen and the app [...]

> So I rebooted my phone and it works again. Not sure how it got into this state but might be worth ruling out a glitch in the matrix? :)

 
@tayloraleach same problem here - now i restarted the phone, it works but the camera crashes right away, but im one step closer i guess!

HerickRaposo, 2021-08-19:
If you use react-native-qrcode-scanner add `checkAndroid6Permissions={true}`  on QRCodeScanner properties definitions and if possible use isFocused. Momentarily this seems to have resolved for me, buit i will continue testing.In my case i tryed call qrscanner in modal components.

## My Modal component:
```
<Modal 
          animationType="slide"
          transparent={true}
          visible={modalVisible}
          onRequestClose={() => {
            setModalVisible(!modalVisible);
          }}
        >
         <QRScannerComponent
            reference={scanner}
            onSucessFunction={onSuccess}
            onPressButtonScan={() => setModalVisible(!modalVisible)}
            typeMarker={isQrScanner ? "QR" : "Barcode"}
          />
        </Modal>
```
## My QRScanner component:
```
import React from "react";
import { TouchableOpacity, Text } from "react-native";
import QRCodeScanner from 'react-native-qrcode-scanner';
import { ButtonWrapper } from '../components/ButtonWrapper';
import styles from '../styles/scanStyle'
import {useIsFocused } from '@react-navigation/native';

interface QRScannerProps {
    reference:any,
    onSucessFunction:any,
    onPressButtonScan:any,
    typeMarker:string
}

export function QRScannerComponent({reference,onSucessFunction,onPressButtonScan,typeMarker}:QRScannerProps){
    let stylemarker;
    stylemarker = (typeMarker == 'Barcode') ? styles.BarcodeMarker : styles.QrMarker
    const isFocused = useIsFocused();
    return(
        isFocused &&
        <QRCodeScanner
            containerStyle={styles.containerStyle}
            cameraStyle={[styles.cameraStyle]}
            markerStyle={stylemarker}
            reactivate={true}
            showMarker={true}
            ref={reference}
            checkAndroid6Permissions={true}
            onRead={onSucessFunction}
            topContent={
                <Text style={{color: '#fff'}}>Please move your camera </Text>
            }
            bottomContent={  
                <TouchableOpacity style={styles.buttonScan} 
                    onPress={onPressButtonScan}>
                    <ButtonWrapper title="Encerrar leitura"/>
                </TouchableOpacity>
            }
        />
    );
}

```

ShivamGoyal26, 2022-11-18:
> After the last few days, we found bugs in our projects. Set `android:hardwareAccelerated="�false"` in `AndroidManifest.xml` makes the camera no longer display. Check your AndroidManifest and set it back to true may help the camera work again.

Perfect thanks a lot solved my problem :)

Brittognanaraj, 2023-02-08:
android:hardwareAccelerated="false" in AndroidManifest.xml makes the camera no longer display. Check your AndroidManifest and set it back to true may help the camera work again.