react-native-camera/react-native-camera #2338: Bundle / APK sizes much bigger after version 2.11.0 with Firebase

pedrosimao, 2019-06-25:
# Question

**Ask your Question**
After updating `react-native-camera` to the latest version `2.11.0` the Android bundle file generates Android `APKs` that are roughly 30% bigger than it used to be with the previous version of the package. And the iOS `IPA` file is more than the double of the size as it used to be. The biggest change I have done was including Firebase TextRecognition SDK in the app. I would like to know if this change alone justifies such a huge difference in app sizes, on both Android and iOS. If someone has any tip on how to make the app smaller, even using the Firebase TextRecognition, this would be a great help!

**Tags**
Firebase, APK, Bundle, IPA, Size, Optimization, Reduce

MateusAndrade, 2019-06-25:
Hi @pedrosimao this is a good question.

Without reading at the docs, i can say to you that its possible since when using TextRecognition you will end add some dependencies to your app. You can read more about that [here](https://firebase.google.com/docs/ml-kit/recognize-text).

Please, let us now If you found that this is related to another reason. 😄

sibelius, 2019-06-25:
for ios you can use cocoapods to only install the dependencies that you want to use

pedrosimao, 2019-06-25:
@sibelius I am already using cocoapods on iOS, before this version I was not. @MateusAndrade I have seen this part of the documentation today: https://firebase.google.com/docs/ml-kit/android/reduce-app-size
Unhappily it seems to talk about Android only, no iOS. And it would be necessary to implement Dynamic Dalivery (https://developer.android.com/studio/projects/dynamic-delivery#dynamic_feature_modules). I will try to dig the information further and I will let you know if I come up with some sort of solution.

andresmtz98, 2019-08-12:
Any update? My app is 10mb larger than previous version (i wasn't using Firebase MLKit).

closed 2019-10-18 by stale[bot]