react-native-camera/react-native-camera #2340: Camera is not working show me white blank screen

Asadullahbarlas, 2019-06-26:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2019-09-01 by stale[bot]

MuhammadUsman786786, 2020-03-16:
Still facing this issue

USAMABARLAS, 2020-03-17:
use react-native image picker, it's working

MuhammadUsman786786, 2020-03-17:
Basically i am using the react native qr code scanner. Which use the react native camera inder the hood

Asadullahbarlas, 2020-03-17:
I think react native image picker is working on qr-scanner