react-native-camera/react-native-camera #2342: Manual focusDepth does not work

liambitjam, 2019-06-26:
**Platform**

Android

**Version**

Android 9

**Description/Current Behaviour**

I am using a slider to adjust the both the zoom and the focus of the camera, the zoom slider works fine but the focus slider does not.

When the slider is at 0 auto-focus is used and works correctly but when the slider is moved from 0 the focusDepth is not being updated. Even when manually setting focusDepth to a value between 0 and 1 there appears to be no change.

**Code**

```
<RNCamera
  ref={cam => {
    this.camera = cam
  }}
  style={styles.preview}
  autoFocus={this.state.focusSliderValue ? RNCamera.Constants.AutoFocus.off : RNCamera.Constants.AutoFocus.on}
  focusDepth={this.state.focusSliderValue ? this.state.focusSliderValue : 0}
  zoom={this.state.zoomSliderValue ? this.state.zoomSliderValue : 0}
>

  <Slider
    style={styles.zoomSlider}
    minimumValue={0}
    maximumValue={1}
    step={0.1}
    minimumTrackTintColor="white"
    maximumTrackTintColor="#7D2D79"
    thumbTintColor="#FFC000"
    value={this.state.zoomSliderValue}
    onValueChange={(zoomSliderValue) => this.setState({ zoomSliderValue })}
  />

  <Slider
    style={styles.focusSlider}
    minimumValue={0}
    maximumValue={1}
    step={0.01}
    minimumTrackTintColor="white"
    maximumTrackTintColor="#7D2D79"
    thumbTintColor="#FFC000"
    value={this.state.focusSliderValue}
    onValueChange={(focusSliderValue) => this.setState({ focusSliderValue })}
  />

</RNCamera>
```

**Expected Behaviour**

focusDepth should be updated by the movement of the focus slider

**Additionals**

I am testing my app using my Nokia 7.1, I have also checked my camera API access using Camera2 Probe for Android 5.0+ and I have the proper focus access.

liambitjam, 2019-07-08:
This image shows the zoom slider at 10% and the focus slider at 0% (auto-focused) both are working as expected

[image]

The next two images then show the focus at manual values of 30% and 100% but there is no difference between the two images.

[image]

[image]

I have manually verified that the values from the slider are being updated, furthermore the zoom slider uses the exact same code and works fine. I have also tried to manually set focusDepth in the RNCamera options but no matter what values are used the focus does not change.

Any help would be much appreciated.

kirill578, 2019-07-19:
I am facing the same issue. I did notice that in some cases pre-setting the focus value does work. but not always

liambitjam, 2019-07-24:
@kirill578 How did you select which camera api you are using?

liambitjam, 2019-09-23:
Is there any advice I could get from the developers as how to fix this problem? The documentation is inaccurate as focusDepth feature does not work.

sibelius, 2019-09-23:
check this https://github.com/react-native-community/react-native-camera/pull/2427

liambitjam, 2019-09-23:
@sibelius thank you very much I will see if this fixes the problem and report back here.

closed 2019-11-29 by stale[bot]

adirrapidapi, 2020-10-24:
@liambitjam did you find solution to manual focus?